AdnanHodzic/auto-cpufreq #604: [FEATURE] Please allow to turn turbo on and off via override

ufechner7, 2023-11-18:
- Searched open issues, this one does not seam to exist yet

Please, add the following commands:
```
sudo auto-cpufreq --force=turbo-boost on
sudo auto-cpufreq --force=turbo-boost off
sudo auto-cpufreq --force=turbo-boost auto
``` 

### System information:
```
auto-cpufreq --debug
```

AdnanHodzic, 2023-11-19:
Yes, having this option/flag could be of great value. @shadeyg56 in GUI on/off/auto could maybe be selected using drop down menu.

Until I or someone else has time to look into developing this feature, if you want to give it a try please feel free to [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

Please note that this is already possible to set by using [auto-cpufreq config file](https://github.com/AdnanHodzic/auto-cpufreq#3-auto-cpufreq-config-file).

```
# turbo boost setting. possible values: always, auto, never
turbo = auto
```

Besides Readme I also cover its functionality as part of [auto-cpufreq v2 video](https://www.youtube.com/watch?v=SPGpkZ0AZVU&t=608s).

shadeyg56, 2023-11-20:
>  GUI on/off/auto could maybe be selected using drop down menu.

Works for me, but needs to be implemented in the CLI first.

I also think it would be better to have it as its own flag instead of using the force flag so something like
`sudo auto-cpufreq --turbo=on`,`sudo auto-cpufreq --turbo=off`, `sudo auto-cpufreq --turbo=auto`

AdnanHodzic, 2023-11-21:
Yes, it should def be its own/new flag.

gelbing, 2023-11-22:
Do you have any idea why it reports that the Ryzen 7 7840U only supports a max frequency of 3300 MHz? I noticed that if using the amd-pstate-epp driver it does report 6600 MHz which is also not correct from my understanding. But it is matching the output of 

## lscpu
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         48 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  16
  On-line CPU(s) list:   0-15
Vendor ID:               AuthenticAMD
  Model name:            AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics
    CPU family:          25
    Model:               116
    Thread(s) per core:  2
    Core(s) per socket:  8
    Socket(s):           1
    Stepping:            1
    Frequency boost:     disabled
    CPU(s) scaling MHz:  38%
    CPU max MHz:         6677.3428
    CPU min MHz:         1600.0000
    BogoMIPS:            6590.08
    ```
    
   Maybe there is something wrong with my understanding

AdnanHodzic, 2023-11-23:
@gelbing please try going thru [Troubleshooting](https://github.com/AdnanHodzic/auto-cpufreq/#troubleshooting) section first, if this doesn't help please open a new issue as it doesn't relate to this feature request.

closed 2024-05-25 by AdnanHodzic

AdnanHodzic, 2025-06-30:
This feature has been implemented as part of #859 which is now merged with `master`.