AdnanHodzic/auto-cpufreq #607: 2.1.0 fails to build inside of a chroot at archlinux

ptr1337, 2023-11-22:
Buildlog:

```
==> Starting build()...

Installing necessary Python packages

/tmp/pkg/src/opt/auto-cpufreq/venv/bin/python
Requirement already satisfied: pip in /tmp/pkg/src/opt/auto-cpufreq/venv/lib/python3.11/site-packages (23.2.1)
Collecting pip
  Obtaining dependency information for pip from https://files.pythonhosted.org/packages/47/6a/453160888fab7c6a432a6e25f8afe6256d0d9f2cbd25971021da6491d899/pip-23.3.1-py3-none-any.whl.metadata
  Downloading pip-23.3.1-py3-none-any.whl.metadata (3.5 kB)
Collecting wheel
  Obtaining dependency information for wheel from https://files.pythonhosted.org/packages/fa/7f/4c07234086edbce4a0a446209dc0cb08a19bb206a3ea53b2f56a403f983b/wheel-0.41.3-py3-none-any.whl.metadata
  Downloading wheel-0.41.3-py3-none-any.whl.metadata (2.2 kB)
Downloading pip-23.3.1-py3-none-any.whl (2.1 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 47.2 MB/s eta 0:00:00
Downloading wheel-0.41.3-py3-none-any.whl (65 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 65.8/65.8 kB 66.6 MB/s eta 0:00:00
Installing collected packages: wheel, pip
  Attempting uninstall: pip
    Found existing installation: pip 23.2.1
    Uninstalling pip-23.2.1:
      Successfully uninstalled pip-23.2.1
Successfully installed pip-23.3.1 wheel-0.41.3

Building auto-cpufreq

Processing /tmp/pkg/src/auto-cpufreq-2.1.0
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [35 lines of output]
      Traceback (most recent call last):
        File "/tmp/pkg/src/opt/auto-cpufreq/venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/tmp/pkg/src/opt/auto-cpufreq/venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pkg/src/opt/auto-cpufreq/venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/poetry/core/masonry/api.py", line 42, in prepare_metadata_for_build_wheel
          poetry = Factory().create_poetry(Path(".").resolve(), with_groups=False)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/poetry_dynamic_versioning/patch.py", line 26, in alt_poetry_create
          name = _get_and_apply_version(
                 ^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/poetry_dynamic_versioning/__init__.py", line 485, in _get_and_apply_version
          version = _get_version(config, name)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/poetry_dynamic_versioning/__init__.py", line 312, in _get_version
          version = _get_version_from_dunamai(vcs, pattern, config)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/poetry_dynamic_versioning/__init__.py", line 264, in _get_version_from_dunamai
          return Version.from_vcs(
                 ^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/dunamai/__init__.py", line 1934, in from_vcs
          return cls._do_vcs_callback(
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/dunamai/__init__.py", line 1973, in _do_vcs_callback
          return callback(**kwargs)
                 ^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/dunamai/__init__.py", line 1066, in from_git
          _detect_vcs(vcs, path)
        File "/tmp/pip-build-env-a0z75_8j/overlay/lib/python3.11/site-packages/dunamai/__init__.py", line 355, in _detect_vcs
          raise RuntimeError(
      RuntimeError: This does not appear to be a Git project
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
==> ERROR: A failure occurred in build().
    Aborting...
mv: cannot stat '/home/notroot/packages/*pkg.tar*': No such file or directory
```

MusicalArtist12, 2023-11-22:
Hey, I pushed 2.1.0-2 onto the AUR, it now builds the virtual environment inside of pacman and adds the flag `POETRY_DYNAMIC_VERSIONING_BYPASS=1` which should fix your issue.

ptr1337, 2023-11-22:
> Hey, I pushed 2.1.0-2 onto the AUR, it now builds the virtual environment inside of pacman and adds the flag `POETRY_DYNAMIC_VERSIONING_BYPASS=1` which should fix your issue.

Thanks, now it worked fine and pushed it to our repository.
Thanks for maintaining it.

closed 2023-11-22 by ptr1337