AdnanHodzic/auto-cpufreq #615: cpu max and min frequency is same after using acpi-freq driver

saikatm, 2023-12-10:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted?  = YES
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? = N/A
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? = YES
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? = YES
---

I want to use the min freq of my cpu, on the default `amd-pstate-epp` driver I was getting 400MHz more or less min freq but after switching to the `acpi-cpufreq `driver I am locked on 1600MHz which will probably drain more battery? please correct me if I am wrong
[image]

---

### System information:

Add/paste output of:

```

Using settings defined in /etc/auto-cpufreq.conf file

-------------------------------------------------------------------------------

Linux distro: Fedora Linux 39 Workstation Edition
Linux kernel: 6.6.4-200.fc39.x86_64
Processor: AMD Ryzen 5 5625U with Radeon Graphics
Cores: 12
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1600 MHz
CPU min frequency: 1600 MHz

Core	Usage	Temperature	Frequency
CPU0      3.0%        44 °C      1329 MHz
CPU1      0.0%        44 °C      1397 MHz
CPU2      2.0%        44 °C      1600 MHz
CPU3      0.0%        44 °C      1600 MHz
CPU4      0.0%        44 °C      1600 MHz
CPU5      1.0%        44 °C      1600 MHz
CPU6      3.0%        44 °C      1464 MHz
CPU7      1.0%        44 °C      1600 MHz
CPU8      1.0%        44 °C      1597 MHz
CPU9      2.0%        44 °C      1395 MHz
CPU10      1.0%        44 °C      1330 MHz
CPU11      0.0%        44 °C      1600 MHz

CPU fan speed: 0 RPM

auto-cpufreq version: 2.1.0 (git: 4dbad55)

Python: 3.12.0
psutil package: 5.9.6
platform package: 1.0.8
click package: 8.1.7
distro package: 1.8.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.4 %

Total CPU usage: 1.3 %
Total system load: 0.37
Average temp. of all cores: 44.00 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

aryan-212, 2023-12-11:
Facing the same issue

AdnanHodzic, 2023-12-11:
Same as #426?

aryan-212, 2023-12-11:
> Same as #426?

Yes.

AdnanHodzic, 2023-12-11:
> > Same as #426?
> 
> Yes.

But #426 is closed, why don't you do the same steps that were done there?

aryan-212, 2023-12-11:
> > > Same as #426?
> > 
> > 
> > Yes.
> 
> But #426 is closed, why don't you do the same steps that were done there?

My bad. #426 his problem was solved after switching to acpi-cpufreq.In our case, it took place after switching to acpi

saikatm, 2023-12-12:
> > > Same as #426?
> > 
> > 
> > Yes.
> 
> But #426 is closed, why don't you do the same steps that were done there?

my is issue is not same as #436 sir.

saikatm, 2023-12-16:
@AdnanHodzic sir help

expl0r3rgu1, 2023-12-18:
[image]
same here with acpi-cpufreq driver

AdnanHodzic, 2025-03-22:
Can same fix which was used in #464 be used here?