AdnanHodzic/auto-cpufreq #623: Ubuntu Install Error

cylemmulo-1, 2024-01-17:
Having a weird issue that I can't find anyone else having the exact problem from what I've seen. It simply just won't install at all. I was on Ubuntu 23, then decided to go back to 22 and it is the exact same problem. I've tried the snap, and it installs but from what I can tell that's not a good way to go. 

### Error output:
```text
installing auto-cpufreq tool

Processing /home/kyle/Downloads/auto-cpufreq-2.1.0
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [26 lines of output]
      Traceback (most recent call last):
        File "/opt/auto-cpufreq/venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/opt/auto-cpufreq/venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/opt/auto-cpufreq/venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/poetry/core/masonry/api.py", line 42, in prepare_metadata_for_build_wheel
          poetry = Factory().create_poetry(Path(".").resolve(), with_groups=False)
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/patch.py", line 26, in alt_poetry_create
          name = _get_and_apply_version(
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 608, in _get_and_apply_version
          version, instance = _get_version(config, name)
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 452, in _get_version
          version = _get_version_from_dunamai(vcs, pattern, config)
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 417, in _get_version_from_dunamai
          return Version.from_vcs(
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1934, in from_vcs
          return cls._do_vcs_callback(
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1973, in _do_vcs_callback
          return callback(**kwargs)
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1066, in from_git
          _detect_vcs(vcs, path)
        File "/tmp/pip-build-env-l4oyfb46/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 355, in _detect_vcs
          raise RuntimeError(
      RuntimeError: This does not appear to be a Git project
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```
---

### System information:

Surface Laptop Go 2
Ubuntu 22.0.4.3
I5-1135G7
8GB memory

hingen, 2024-01-20:
Did you install via the instructions stated in the [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq#auto-cpufreq-installer) section of the README? If not could you perhaps list down the commands which you ran?

rootCircle, 2024-01-21:
> Wondering if it could be cause by the repo not being obtained via git clone

Yes, it is. The project requires `poetry_dynamic_versioning`, which requires that `.git` folder is present at the time of build. Which wasn't I believe in this case.

@cylemmulo-1 You can simply use `git clone https://github.com/AdnanHodzic/auto-cpufreq.git` instead of manually downloading it from zip and rest installation instruction remains the same. (This needs to be updated in README too)

AdnanHodzic, 2024-01-21:
> This needs to be updated in README too

Could someone please do this?

rootCircle, 2024-01-21:
I will do it! :smile:

closed 2024-01-21 by AdnanHodzic

AdnanHodzic, 2024-01-21:
Issue was automatically closed #623 PR merge, as [ auto-cpufreq README](https://github.com/AdnanHodzic/auto-cpufreq/) has been updated.

cylemmulo-1, 2024-01-22:
Thanks a lot guys! Yeah the git clone solved it! Appreciate the help

AdnanHodzic, 2024-01-22:
Most welcome & please consider [donating or contributing to the project](https://github.com/AdnanHodzic/auto-cpufreq/#donate)!

PR #630 (2024-01-21): docs(README): add instruction for failing buildings in case of missing git directory
Builds when done through external zip file having source code, won't work out and will result in error due to `poetry_dynamic_versioning` requiring .git directory. This PR seem to add specific instruction regarding that in README

Fixes #623