AdnanHodzic/auto-cpufreq #639: `scaling_available_frequencies`: No such file or directory

sphh, 2024-02-06:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? : Yes
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? : n/a
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? : n/a
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? : n/a

### Error output:
```sh
$ cpufreqctl 
CPU driver: intel_cpufreq
Governors: conservative ondemand userspace powersave performance schedutil
cat: /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies: No such file or directory
Frequencies: 

$ ls /sys/devices/system/cpu/cpu0/cpufreq/
affected_cpus               related_cpus                 scaling_governor
cpuinfo_max_freq            scaling_available_governors  scaling_max_freq
cpuinfo_min_freq            scaling_cur_freq             scaling_min_freq
cpuinfo_transition_latency  scaling_driver               scaling_setspeed

$ cat /sys/devices/system/cpu/cpu0/cpufreq/*_max_freq
3300000
3300000

$ cat /sys/devices/system/cpu/cpu0/cpufreq/*_min_freq
800000
800000
```
---

### System information:

Add/paste output of:

```
$ auto-cpufreq --stats

		Executed on: February 06 (Tuesday) - 12:44:38

-------------------------------------------------------------------------------

Linux distro: Linux Mint 21.3 Virginia
Linux kernel: 6.7.2-surface-1
Processor: Intel(R) Core(TM) i7-4650U CPU @ 1.70GHz
Cores: 4
Architecture: x86_64
Driver: intel_cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3300 MHz
CPU min frequency: 800 MHz

Core	Usage	Temperature	Frequency
CPU0     26.8%        70 °C       898 MHz
CPU1     28.6%        63 °C       898 MHz
CPU2     32.3%        70 °C       898 MHz
CPU3     25.0%        63 °C       898 MHz

---------------------------- CPU frequency scaling ----------------------------

Battery is: charging

Setting to use: "conservative" governor
Not setting EPP (not supported by system)

Total CPU usage: 50.6 %
Total system load: 1.74
Average temp. of all cores: 66.50 °C 

High CPU load (load average: 1.74, 2.05, 1.98)
setting turbo boost: on

-------------------------------------------------------------------------------
```
I use a kernel from https://github.com/linux-surface/linux-surface/

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.
---

- Call `cpufreqctl` without any parameters.
- Get error as described above.

Is this due to the special flavour of kernel or has this file been removed from the `/sys/` filesystem?

AdnanHodzic, 2024-02-07:
Have you tried using different driver, i.: `intel_pstate`? https://github.com/AdnanHodzic/auto-cpufreq?tab=readme-ov-file#troubleshooting

sphh, 2024-02-07:
Actually no, because none of the cases mentioned seemed to be the problem I encountered.

Now I started Linux with the `intel_pstate=disabled` command line. This removes the error message:
```sh
$ cpufreqctl
CPU driver: acpi-cpufreq
Governors: conservative ondemand userspace powersave performance schedutil
Frequencies: 2301000 2300000 2200000 2100000 2000000 1800000 1700000 1600000 1500000 1400000 1300000 1200000 1000000 900000 800000 759000
```

Thanks.

PS: You might want to add this case to the reasons, why you might want to change the driver used …

closed 2024-02-07 by sphh