AdnanHodzic/auto-cpufreq #641: issue w.r.t --monitor and --live flag

madmax5647, 2024-02-09:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? Yes
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? 
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? Yes
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? yes

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---
this is when running auto-cpufreq with --monitor flag

[image]

the above one is when i was not using a config file.
but when i created a config file as mentioned in the document i am getting the below result

[image]

also, when running auto-cpufreq with --live flag, i am getting this write error. but only when plugged in(using performance governor and balance performance epp)

[image]

i found a thread w.r.t --live flag. but it was with amd processors. did'nt know what to do with that. new to linux

also, installed it via git repo

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Linux distro: Fedora Linux 39 Thirty Nine
Linux kernel: 6.7.3-200.fc39.x86_64
Processor: Intel(R) Core(TM) i5-10210U CPU @ 1.60GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1600 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0      0.0%        37 °C      1600 MHz
CPU1      0.0%        38 °C       400 MHz
CPU2      0.0%        40 °C       400 MHz
CPU3      1.0%        38 °C      1600 MHz
CPU4      1.0%        37 °C      1600 MHz
CPU5      2.9%        38 °C      1600 MHz
CPU6      0.0%        40 °C      1600 MHz
CPU7      1.0%        38 °C      1601 MHz

auto-cpufreq version: 2.1.0 (git: 3e8ee20)

Python: 3.12.1
psutil package: 5.9.8
platform package: 1.0.8
click package: 8.1.7
distro package: 1.9.0

Computer type: Convertible
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.38 %

Total CPU usage: 0.3 %
Total system load: 0.06
Average temp. of all cores: 38.25 °C

Currently using: performance governor
Currently turbo boost is: off

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

Thanks in advance

PurpleWazard, 2024-02-09:
could i have your /etc/auto-cpufreq.conf file please

PurpleWazard, 2024-02-09:
i believe Ive found the issue. i will submit a pull request.

madmax5647, 2024-02-10:
> could i have your /etc/auto-cpufreq.conf file please im working on the battery stuff for this project and having your config file would help me.

```

# settings for when connected to a power source
[charger]
# see available governors by running: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
# preferred governor
governor = performance

# EPP: see available preferences by running: cat /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences
energy_performance_preference = balanced_performance

# minimum cpu frequency (in kHz)
# example: for 800 MHz = 800000 kHz --> scaling_min_freq = 800000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_min_freq = 800000

# maximum cpu frequency (in kHz)
# example: for 1GHz = 1000 MHz = 1000000 kHz -> scaling_max_freq = 1000000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_max_freq = 1000000

# turbo boost setting. possible values: always, auto, never
turbo = auto

# settings for when using battery power
[battery]
# see available governors by running: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
# preferred governor
governor = powersave

# EPP: see available preferences by running: cat /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences
energy_performance_preference = balanced_performance

# minimum cpu frequency (in kHz)
# example: for 800 MHz = 800000 kHz --> scaling_min_freq = 800000

```

i changed only the epp options.

madmax5647, 2024-02-10:
@PurpleWazard i updated my battery.py script with the code you provided manually and now --monitor flag is not throwing any exceptions. but the --live flag stills throws the exception. i wonder if i could ignore it if it does not affect the working itself. because it does not throw any error while running with --stats flag after installing the daemon

PurpleWazard, 2024-02-10:
The battery scripts don't effect --live I'm not having any issue with --live so I'm not sure what it is.

AdnanHodzic, 2024-02-10:
@madmax5647 curios, after you pull the latest changes if you change to some [other (intel) driver](https://github.com/AdnanHodzic/auto-cpufreq/#troubleshooting) will you get the same error?

closed 2024-02-10 by AdnanHodzic

madmax5647, 2024-02-10:
@AdnanHodzic No i am not getting any error. but the epp setting is not applied anymore. it is showing "Not setting EPP (not supported by system)".I switched to acpi-cpufreq

this is when i run with --live flag
```

Linux distro: Fedora Linux 39 Thirty Nine
Linux kernel: 6.7.3-200.fc39.x86_64
Processor: Intel(R) Core(TM) i5-10210U CPU @ 1.60GHz
Cores: 8
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1600 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0      2.0%        40 °C      1600 MHz
CPU1      1.0%        41 °C      1601 MHz
CPU2      2.0%        42 °C      1600 MHz
CPU3      1.0%        41 °C      1600 MHz
CPU4      1.0%        40 °C      1600 MHz
CPU5      1.0%        41 °C      1602 MHz
CPU6      2.0%        42 °C      1600 MHz
CPU7      2.0%        41 °C      1600 MHz

---------------------------- CPU frequency scaling ----------------------------

Battery is: charging

Setting to use: "performance" governor
Not setting EPP (not supported by system)

Total CPU usage: 1.1 %
Total system load: 0.16
Average temp. of all cores: 41.00 °C

Load optimal (load average: 0.16, 0.19, 0.10)
setting turbo boost: off

```

AdnanHodzic, 2024-02-12:
Sorry I wasn't able to reproduce this either, everything works fine for me with `--live`. 

@PurpleWazard but for some weird reason I'm not seeing EPP at all in `--monitor`

commit 005b4aa1 (2024-02-10): enable_tresholds config file improvements (Closes: #641)