AdnanHodzic/auto-cpufreq #647: Battery conservation_mode?

fridrichgood, 2024-02-15:
I don't know if battery charge thresholds limit works on my laptop, I have lenovo legion and in modules I have ideapad_laptop instead of ideapad_acpi. But i have battery conservation mode and can switch it here:

echo 1 > /sys/bus/platform/drivers/ideapad_acpi/VPC2004\:00/conservation_mode

Is it possible to add this toggle to config file and cli key for switching? TLP had this option.

PurpleWazard, 2024-02-15:
yea i could do that. however i wont be able to test it so ill need your help with that.

shadeyg56, 2024-02-15:
[Here](https://wiki.archlinux.org/title/Laptop/Lenovo#Battery_conservation_mode) is a link to the Arch wiki that has a good explanation of Conservation mode

fridrichgood, 2024-02-15:
> yea i could do that. however i wont be able to test it so ill need your help with that.

Ok, i will be happy to help.

PurpleWazard, 2024-02-18:
hey so it took a little longer than i thought. here is my repo https://github.com/PurpleWazard/auto-cpufreq/tree/master do you mind trying it out?

to install run auto-cpufreq-installer with sudo
run tests with sudo auto-cpufreq --live or --monitor 
in /etc/auto-cpufreq.conf add the options under [battery]

test 1 
#enable_thresholds = true
#start_threshold = 20
#stop_threshold = 80
#ideapad_laptop_conservation_mode = true

test 2
enable_thresholds = true
#start_threshold = 20
#stop_threshold = 80
ideapad_laptop_conservation_mode = true

test 3
enable_thresholds = true
start_threshold = 20
stop_threshold = 80
ideapad_laptop_conservation_mode = true

test 4
enable_thresholds = true
start_threshold = 20
stop_threshold = 80
ideapad_laptop_conservation_mode = false

fridrichgood, 2024-02-20:
> Custom threshold setting not working for me on Lenovo Ideapad. I guess only the default conservation mode value of 60 is supported?

Yes. Battery stops charging around 58-60%

fridrichgood, 2024-02-20:
> hey so it took a little longer than i thought. here is my repo https://github.com/PurpleWazard/auto-cpufreq/tree/master do you mind trying it out?
>
> to install run auto-cpufreq-installer with sudo [...]

Conservation mode works as expected. Thresholds ignored, battery charging above stop_threshold, but i think it normal for my hardware.

fridrichgood, 2024-02-20:
@PurpleWazard Thank you for your work. I close this issue.

closed 2024-02-20 by fridrichgood