AdnanHodzic/auto-cpufreq #654: Disabling CPU cores crashes the daemon.

shadeyg56, 2024-02-22:
This bug was discovered in #603. 
Running `echo 0 | sudo tee /sys/devices/system/cpu/cpuN/online` (where N is any integer greater than 0) causes the daemon to crash

This can be reproduced by running the command then running `systemctl status auto-cpufreq` or `auto-cpufreq --stats`

I'm going to look into this and hopefully open a PR to resolve this issue.

shadeyg56, 2024-02-22:
Unfortunately, this looks like an upstream bug in `psutil`
https://github.com/giampaolo/psutil/issues/2254

So this isn't something we can fix. The issue on the psutil repo has had little activity as well. If I have some time, I might delve into the psutil source code and see if I can open a PR over there

shadeyg56, 2024-02-22:
Ok i've submitted a PR on the psutil repo that should fix this.
https://github.com/giampaolo/psutil/pull/2376

I imagine it will take some time for this to merge and make it into a release so we just have to wait.

AdnanHodzic, 2024-03-03:
Congrats on getting it merged!

astonish-g, 2024-03-12:
Thank you so much beeing nice enough to fix this issue! Now I changed my laptop, and have one with AMD Ryzen 8845HS. I will try this, but for now I am getting higher watt useage compared to my old one. I am still trying to find the issue, before I will try this fix. Thank you again!

Svenum, 2024-03-30:
Hi,
I do not know if this is the right place, but I do not want to open an new issue only for this question. 
Is there a way to set CPU-Core-Count via the config file? I want to set my core count to 4 if on battery and to 16 if on AC.

shadeyg56, 2024-04-01:
Its not currently possible. 
1. because it hasn't been implemented
2. this bug is still in upstream as the merged PR has not made its way into a release yet so disabling a cpu core will crash auto-cpufreq

Nolicx, 2024-04-09:
Hey there, i've been using this tool for about the past week and i think it's great.
While searching for a way to disable cores i've found this issue and seen a few other issues and discussions regariding this feature.
I would love to implement it, as soon as the upstream issue has been fixed. 
Is there anything i should do besides the implementation and the creation of a pull request?

shadeyg56, 2024-04-09:
I'm glad auto-cpufreq has been useful for you.

If you're interested in adding it, I think its a good idea and shouldn't be too difficult.
Although we don't know when a new release for psutil will happen, it should be possible to swap the PyPi release directly with the Github source  (from the commit of the patch I made) in the meantime
Consider joining our [Discord server](https://discord.com/invite/Sjauxtj6kH) if you want to discuss anything.

shadeyg56, 2024-04-10:
>Although we don't know when a new release for psutil will happen, it should be possible to swap the PyPi release directly with the Github source (from the commit of the patch I made) in the meantime

Went ahead and did this. 5e17ee3

commit 5e17ee31 (2024-04-10): update psutil to patched Git version
Fixes #654
closed 2024-04-10 by shadeyg56