AdnanHodzic/auto-cpufreq #681: performance mode not working like a performance

gnuchanos, 2024-04-23:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?
[image]
[image]
still working low cpu frequency why? why i can't 2.3 ghz?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

AdnanHodzic, 2024-05-25:
What exactly did you mean to say in this issue or what is the problem you're facing?

gnuchanos, 2024-05-25:
If you look at the screenshot, you can understand what the problem is, but I removed this from my system loo late

closed 2024-05-25 by gnuchanos