AdnanHodzic/auto-cpufreq #685: auto-cpufreq locks up on Fedora/Gnome

fonnesbeck, 2024-04-30:
I'm getting frequent lockups when running on Fedora 40 and Gnome 46. Seems to happen occasionally after wake from sleep (or at least that is when I notice it).

### Error output:

[image]

### System information:

You can see the system information in the corner of the screen shot above. Will try running again with debug turned on and add more info.

---

AdnanHodzic, 2024-04-30:
@shadeyg56 now that it's brought up I also experience  this on Ubuntu. Can you reproduce?

@fonnesbeck silverlining is that only GUI component freezes, daemon continues running in background. You can verify by running `auto-cpufreq -stats` in Terminal next time this happens.

In meantime since template issue was ignored and wasn't filled out, making this issue invalid for project developers or I to act on.

Hence, please properly filled out requested fields.

Reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

fonnesbeck, 2024-04-30:
Apologies, added template fields to report.

shadeyg56, 2024-04-30:
I'm unable to reproduce this on NixOS. I installed and tested on GNOME as well, on both X11 and Wayland. The GNOME version on Nix is currently 45.5 so maybe its a GNOME 46 issue? Does this issue occur on any other desktop environments?

AdnanHodzic, 2024-05-01:
> The GNOME version on Nix is currently 45.5 so maybe its a GNOME 46 issue?

No, I had this problem with previous GNOME version (wayland)s, since the release of v2.0 that introduced GUI component, I just never bothered reporting it.

Unless we had logging for `auto-cpufreq-gtk`, which I just ran in terminal to see if it'll return anything once it freezes, only thing we can do is [what I suggested in 677](https://github.com/AdnanHodzic/auto-cpufreq/issues/677#issuecomment-2067655626) to try catching what happens excatly.

AdnanHodzic, 2024-05-07:
> Until we had logging for `auto-cpufreq-gtk`, which I just ran in terminal to see if it'll return anything once it freezes

Only output I got from `auto-cpufreq-gtk` after running it in terminal after suspend is: `Killed`. Please note, this will only happen if you leave it running after extended period of time.

Heavygrass, 2024-10-14:
Hi Guys,
I also experience GUI freeze on Cinnamon 6.0.4 (Ubuntu 24.04.1 LTS), which usually starts every time after a few minutes of GUI restart. 
E.g. "Governor override" horizontal buttons stop working and cores usage is frozen.
After restart GUI usually works for about a minute or so.
grep auto-cpufreq-gtk /var/log/syslog 
gives nothing.

Let me suggest to **add an option to GUI to generate verbose log**, so that we could share it here to help developers.