AdnanHodzic/auto-cpufreq #688: Wake up and reboot problems

RaffaeleMelluso, 2024-05-07:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted?  Yes
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? Installed Correctly
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? Yes
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? Yes

### Error output:
```text
No error showned, 
```
---

### System information:

Add/paste output of

```
auto-cpufreq --debug
```
CPU max frequency: 2800 MHz
CPU min frequency: 900 MHz

Core	Usage	Temperature	Frequency
CPU0      0.0%        40 °C      1927 MHz
CPU1      0.0%        40 °C       900 MHz
CPU2      2.0%        40 °C      1821 MHz
CPU3      0.0%        40 °C      3025 MHz
CPU4      1.0%        40 °C       900 MHz
CPU5      1.0%        40 °C      1930 MHz
CPU6      1.0%        40 °C      4341 MHz
CPU7      0.0%        40 °C      2800 MHz

auto-cpufreq version: 2.2.0 (git: 0815e7e)

Python: 3.12.3
psutil package: 5.9.8
platform package: 1.0.8
click package: 8.1.7
distro package: 1.9.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.41 %

Total CPU usage: 0.6 %
Total system load: 0.42
Average temp. of all cores: 40.00 °C 

Currently using: schedutil governor
Currently turbo boost is: on

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.
Just a black screen light when i try to reboot or wake up my system. When i shutted of cpuautofreq it started working again.

---

shadeyg56, 2024-05-07:
You'll need to provide more information, as I've never heard of anything like this happening.

AdnanHodzic, 2024-05-08:
> You'll need to provide more information, as I've never heard of anything like this happening.

Also not sure how it could be related to auto-cpufreq ... as it has no impact on Wayland/X11 which might be a problem here.

AdnanHodzic, 2024-05-08:
Glad to hear you figured it out, closing the issue!

closed 2024-05-08 by AdnanHodzic