AdnanHodzic/auto-cpufreq #699: AUR dependency error

CRAG666, 2024-05-13:
Traceback (most recent call last):
  File "/usr/bin/auto-cpufreq", line 5, in <module>
    from auto_cpufreq.bin.auto_cpufreq import main
  File "/usr/lib/python3.12/site-packages/auto_cpufreq/bin/auto_cpufreq.py", line 14, in <module>
    from auto_cpufreq.core import *
  File "/usr/lib/python3.12/site-packages/auto_cpufreq/core.py", line 29, in <module>
    from auto_cpufreq.utils.config import config
  File "/usr/lib/python3.12/site-packages/auto_cpufreq/utils/config.py", line 2, in <module>
    from auto_cpufreq.utils.config_event_handler import ConfigEventHandler
  File "/usr/lib/python3.12/site-packages/auto_cpufreq/utils/config_event_handler.py", line 1, in <module>
    import pyinotify
ModuleNotFoundError: No module named 'pyinotify'

Ab3loth, 2024-05-13:
paru -S python-pyinotify

needs to be added as a dependency to the aur package at some point as well but that's whats missing

shadeyg56, 2024-05-13:
@parmjotsinghrobot @MusicalArtist12 could one of you guys help with this?

MusicalArtist12, 2024-05-13:
Yeah we need to add python-pyinotify as a dependency

MusicalArtist12, 2024-05-13:
`pacman -S python-pyinotify`

closed 2024-05-14 by CRAG666