AdnanHodzic/auto-cpufreq #700: [Ideapad] `--stats` gives error as `BAT0` can't be found

LnLcFlx, 2024-05-14:
### Error output of `auto-cpufreq --stats`:
```text

Using settings defined in /etc/auto-cpufreq.conf file

Note: You can quit stats mode by pressing "ctrl+c"
could not get value from conservation mode

-------------------------------- Battery Info ---------------------------------

battery count = 1
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/auto_cpufreq/battery_scripts/ideapad_laptop.py", line 92, in ideapad_laptop_print_thresholds
    with open(f'/sys/class/power_supply/BAT{b}/charge_start_threshold', 'r') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/power_supply/BAT0/charge_start_threshold'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/sbin/auto-cpufreq", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/auto_cpufreq/bin/auto_cpufreq.py", line 157, in main
    battery_get_thresholds()
  File "/usr/lib/python3.12/site-packages/auto_cpufreq/battery_scripts/battery.py", line 42, in battery_get_thresholds
    ideapad_laptop_print_thresholds()
  File "/usr/lib/python3.12/site-packages/auto_cpufreq/battery_scripts/ideapad_laptop.py", line 101, in ideapad_laptop_print_thresholds
    print(f"ERROR: failed to read battery thresholds: {e}")
                                                       ^
NameError: name 'e' is not defined
```

### System information:
I'm running Arch on an IdeaPad 5 Pro 14ACN6.
My battery is indeed on `BAT1` and not `BAT0` but I don't know of an option to specify this.

AdnanHodzic, 2024-05-14:
@PurpleWazard ^

@LnLcFlx if you want us to help you with this problem, please fill out the issue template. 

Reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

LnLcFlx, 2024-05-14:
I know and I have tried all the above and running `--debug` gives the same error ironically... thats why I left it out

shadeyg56, 2024-05-14:
I'm at work, but I can try to push out a patch for this later, if Purple can't get to it

shadeyg56, 2024-05-14:
[1f911b5](https://github.com/AdnanHodzic/auto-cpufreq/commit/1f911b534c078e41d02fd78decbac5dc6f3cc4ec)

Should be fixed, but since I don't have any supported laptops, I couldn't test for myself.
Give it a try please!

closed 2024-05-14 by shadeyg56