AdnanHodzic/auto-cpufreq #720: [ Nix / NixOS ] PR #718 breaks flake

cig0, 2024-06-03:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
error:
       … while calling the 'seq' builtin

         at /nix/store/i1aw9jjgxcvyd642s12kw3iasmarwd42-source/lib/modules.nix:322:18:

          321|         options = checked options;
          322|         config = checked (removeAttrs config [ "_module" ]);
             |                  ^
          323|         _module = checked (config._module);

       … while evaluating a branch condition

         at /nix/store/i1aw9jjgxcvyd642s12kw3iasmarwd42-source/lib/modules.nix:261:9:

          260|       checkUnmatched =
          261|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
             |         ^
          262|           let

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: undefined variable 'mkForce'

       at /nix/store/f641gq84hc4hka15bln4wnqdkxg4v49a-source/nix/module.nix:47:35:

           46|         serviceConfig.WorkingDirectory = "";
           47|         serviceConfig.ExecStart = mkForce [
             |                                   ^
           48|           ""

```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

Hi, PR https://github.com/AdnanHodzic/auto-cpufreq/pull/718 is broken for me (see error above); I had to revert to the previous commit:

```nix
    auto-cpufreq = { # Energy efficiency
      inputs.nixpkgs.follows = "nixpkgs";
      url = "github:AdnanHodzic/auto-cpufreq?rev=3f6d7a3e77732c1dbe4873b69404fefd899ad35e";
    };
```

shadeyg56, 2024-06-04:
1e44c80

Sorry about that

closed 2024-06-04 by shadeyg56

isabelroses, 2024-06-05:
Ahh sorry, thats my bad, should've caught that.