AdnanHodzic/auto-cpufreq #726: Sleep mode

Andrey446, 2024-06-20:
Every time I close the laptop, it goes into sleep mode, although I set it to ignore in the config in Fedora 40. But since I use excellent software in the good sense of this understanding, I do not know where to disable this feature.

AdnanHodzic, 2024-06-23:
Sorry, but this has nothing to do with auto-cpufreq, so closing the issue. 

Try giving [this Google search query a try ](https://www.google.com/search?q=disable+suspend+on+lid+close+fedora&oq=disable+suspend+on+lid+close+fedora)...

closed 2024-06-23 by AdnanHodzic

Andrey446, 2024-06-23:
How does it not apply if your software replaces the power settings thereby maintaining the same sleep mode after removing it, everything becomes normal

AdnanHodzic, 2024-06-23:
Then don't ignore the issue template when submitting a new issue, if you want me or somone else to look into the problem you're facing. Feel free to re-open this issue with properly filled out requested fields.

Reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behavior.

shadeyg56, 2024-06-23:
`auto-cpufreq` does not touch system settings, so yes please provide more info if you really believe this is related to `auto-cpufreq`