AdnanHodzic/auto-cpufreq #727: [Feature Request] Force/override EPP

nghiapq77, 2024-06-29:
Similar to forcing governor with `auto-cpufreq --force powersave/performance`.
Would be nice to know if this is feasible. Thank you.

AdnanHodzic, 2024-06-29:
You're requesting same way you could force governor to be in i,e "performance". You would like to have ability to only force "EPP" the same way?

nghiapq77, 2024-06-29:
Yes. On my machine, forcing governor to powersave on AC would also force EPP to value from battery config.
[image]

A4Alpha, 2024-07-01:
> Yes. On my machine, forcing governor to powersave on AC would also force EPP to value from battery config. ![pic-selected-240629-1651-40](https://private-user-images.githubusercontent.com/28785377/344383365-459208c3-6302-4314-a4da-bbcb9b882c38.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3OTMxNjUsIm5iZiI6MTcxOTc5Mjg2NSwicGF0aCI6Ii8yODc4NTM3Ny8zNDQzODMzNjUtNDU5MjA4YzMtNjMwMi00MzE0LWE0ZGEtYmJjYjliODgyYzM4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAxVDAwMTQyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFlNGNlMDk4NjY1N2FlZDBiZThiZGRmZTA2NWVmZGE1NzNjM2FkNzQ1NzUyZWZhYzEzOWY2NzZjZjBkZmE5MWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.izzXzjW5vfvDcVq6XzrHACeS3QkwMw12EMn4wBlnyFg)

Sorry, off topic, Can I know the font you're using? @nghiapq77

nghiapq77, 2024-07-01:
> Sorry, off topic, Can I know the font you're using? @nghiapq77

ttf-monaco on AUR.