AdnanHodzic/auto-cpufreq #728: "update" parameter doesn't exist anymore

yoyonel, 2024-07-01:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Description

https://github.com/AdnanHodzic/auto-cpufreq?tab=readme-ov-file#update---auto-cpufreq-update
[image]

### Error output:
```shell
╰─ sudo auto-cpufreq --version

-------------------------------------------------------------------------------

Linux distro: Debian GNU/Linux 11 bullseye
Linux kernel: 5.10.0-30-amd64
auto-cpufreq version:
Git commit: v2.2-606-gec1f1cd3d

-------------------------------------------------------------------------------

╰─ sudo auto-cpufreq --update
Usage: auto-cpufreq [OPTIONS]
Try 'auto-cpufreq --help' for help.

Error: No such option: --update Did you mean --daemon?
```
---

AdnanHodzic, 2024-07-06:
Wasn't able to replicate this problem:

```
sudo auto-cpufreq --update
auto-cpufreq is up to date
```

Will also ask on [Discord community](https://discord.gg/Sjauxtj6kH) if anyone else was able to replicate it.

Angel-Karasu, 2024-07-13:
```shell
auto-cpufreq --version

-------------------------------------------------------------------------------

Linux distro: Artix Linux  
Linux kernel: 6.9.7-artix1-1
auto-cpufreq version: 2.3.0 (git: f300d31)

-------------------------------------------------------------------------------

sudo auto-cpufreq --update
auto-cpufreq is up to date
```

AdnanHodzic, 2024-07-14:
> > Wasn't able to replicate this problem:
> > ```
> > sudo auto-cpufreq --update
> > auto-cpufreq is up to date
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Will also ask on [Discord community](https://discord.gg/Sjauxtj6kH) if anyone else was able to replicate it.
> 
> Thx, can you provide the result of `sudo auto-cpufreq --version` please ?

```
auto-cpufreq --version 

-------------------------------------------------------------------------------

Linux distro: Ubuntu 24.04 Noble Numbat
Linux kernel: 6.8.0-38-generic
auto-cpufreq version: 2.3.0 (git: f300d31)

-------------------------------------------------------------------------------
```

closed 2025-03-22 by AdnanHodzic