AdnanHodzic/auto-cpufreq #729: Tray icon

yusifvatan, 2024-07-02:
Hi. It would be great add the tray icon of application. It would be easy to change power modes

AdnanHodzic, 2024-07-06:
This is currently only possible with [auto-cpufreq GUI](https://github.com/AdnanHodzic/auto-cpufreq#install---auto-cpufreq-daemon). 

If you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

lukasnxyz, 2024-11-02:
I will have a look at this.

AdnanHodzic, 2025-06-01:
@EBendinelli this is amazing!

On first look extension is definitively rough around the edges and some things could be improved, but in a nutshell, it works! For 4 years [I had the same idea so it's great to see someone actually did it.](https://github.com/AdnanHodzic/auto-cpufreq/discussions/312)

> In its current form auto-cpufreq is very stable and besides implementing [existing feature requests](https://github.com/AdnanHodzic/auto-cpufreq/issues?q=is%3Aissue+feature+is%3Aopen), next big feature I would need help with is to have it implemented as a [GNOME extension](https://extensions.gnome.org/).

Would you be interested in [creating a PR to work towards merging this with `master` branch?](https://github.com/AdnanHodzic/auto-cpufreq#code-contribution). This would be a great feature and something that would be a highlight of auto-cpufreq 3.0 release.

EBendinelli, 2025-06-01:
> definitively rough around the edges
I didn't expect anything less after many years without coding haha! But glad to hear it works :)

Happy to have a look into making a proper PR but you'll have to tell me how you envision the integration with the main codebase. 

Do you see this integrated as an option to install/enable the extension? Something in the config? Or simply have the extension's code in this repo, publish on https://extensions.gnome.org/ and keep separate from the main app from a workflow perspective?

AdnanHodzic, 2025-06-02:
> Happy to have a look into making a proper PR but you'll have to tell me how you envision the integration with the main codebase.

Great!

> Do you see this integrated as an option to install/enable the extension? 
> Something in the config? Or simply have the extension's code in this repo, publish on https://extensions.gnome.org/ and keep separate from the main app from a workflow perspective?

If need be we could have it as an additional flag which could install the extension. 

But not sure if it would add any value, I think it would be better to simply have the extension published on https://extensions.gnome.org, as that would also simplify README changes. 

Regarding the rest, we could simply add rest of the extension code in auto-cpufreq repo and put it into e.g "gnome-extension" dir. If there are a lot of changes from your side, I wouldn't mind give you collaborator access, so you could push changes without me having to approve every PR.

In beginning, until extension is in state we want it in, all changes could be made on a separate branch before we merge it with `master`.

Let me know, and if you want to discuss this more in detail I also pop up [auto-cpufreq Discord server](https://discord.gg/Sjauxtj6kH) from time to time. Otherwise Github also works fine :)

EBendinelli, 2025-06-14:
So I've been daily driving this and so far so good. The only issue is I recently updated to 2.6.0 (was using 2.5.0) and the hack I was using to identify the governor on start doesn't work anymore (I was looking for the "Warning" keyword to distinguish between default/balanced and manually set governor).

As far as I can tell right now there is no way to distinguish what governor is selected with the `--stats` command anymore. It returns the governor set, whether it was set by the user or the balanced/default mode.

So I guess we may need a very simple `--status` command to get the status which this extension could use. With that I could also add some code to check the status at regular interval so that if the mode is switched from the GUI the extension updates its status (this currently doesn't work).

If you want to give me so suggestions of where to look at, I could try and implement this. Also open to suggestions if you have a better idea !

Also a note that in the GUI, discarding the password prompt when trying to switch to a different governor will result in the governor radio button being enabled (e.g: click powersave > click cancel on password prompt > powersave radio button is enabled).

JuanR4140, 2025-07-02:
@AdnanHodzic @EBendinelli Seeing as a solution is being worked on for GNOME, I decided to give it a try and do something of a tray icon for KDE Plasma. It's not perfect and has some small kinks that need to be worked out (possibly width/height sizing and mouse clicks outside the widget) but overall details some information from the log at a set 2.5s interval and can also override settings, similar to the GUI. Attached is an image of the widget (I couldn't find an icon file for auo-cpufreq so it's a placeholder) working. I haven't published the code either as it requires the folder/metadata to be named a specific way that I'm not sure would be best/approved by the team. Let me know your thoughts/suggestions, if it's any good, etc. Should this be looked into further, I'd like to be able to contribute to the process of getting things integrated right. Cheers!

[image]

JuanR4140, 2025-07-02:
Would there be any issue with leaving this choice to the user in the install script (Either in the auto-cpu-freq-installer or auo-cpu-freq --install)? I was thinking the extension could be stored in its own gnome-extension-folder, and in the install script you could detect what DE the user is in, maybe with the `$XDG_CURRENT_DESKTOP` variable, and if its GNOME, ask the user if they'd like to install the extension. Of couse, once they confirm, you can `cp` the extension to the right location, enable it and all is well, maybe. Just an idea..

JuanR4140, 2025-07-02:
For CLI installs through the auto-cpufreq-installer, would there be any issues with leaving that choice up to the user? Maybe you can detect what DE the user is in with the `$XDG_CURRENT_DESKTOP` variable and ask them if they'd like to install the extension. If they confirm, you can `cp` the extension from the folder to the right location and enable it. I think that would work for manual installation. For package managers though, that would have to be different. Just an idea..

AdnanHodzic, 2025-07-05:
Syncing in parts of [communication that happened on Discord](https://discord.com/channels/1158239366251020361/1158239367337349164/1390926233394483285).

In summary, once it's all completed, I think it would the best to update README (somewhere towards the top) with something like:

```
### 🌐 Community Projects 

If you're using auto‑cpufreq in desktop environments, you may find these community integrations helpful and useful:

- GNOME Extension: [auto-cpufreq-gnome-extension](https://github.com/example/auto-cpufreq-gnome-extension)
- KDE Widget: [auto-cpufreq-kde-widget](https://github.com/example/auto-cpufreq-kde-widget)
```