AdnanHodzic/auto-cpufreq #754: [ NixOS ] FileNotFoundError: [Errno 2] No such file or directory: 'lsmod'

cig0, 2024-07-29:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
journalctl -xeu auto-cpufreq.service
░░ A start job for unit auto-cpufreq.service has finished successfully.
░░ 
░░ The job identifier is 8006.
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]: Traceback (most recent call last):
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/yf5x4z7yir309qd3qyyn8w06kpxcfzxj-python3.11-auto-cpufreq-2.3.0/bin/..auto-cpufreq-wrapped-wrapped", line 9, in <module>
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     sys.exit(main())
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:              ^^^^^^
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/43pnqjci11b8rki32vvybms8f1k6v0rw-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     return self.main(*args, **kwargs)
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/43pnqjci11b8rki32vvybms8f1k6v0rw-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 1078, in main
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     rv = self.invoke(ctx)
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:          ^^^^^^^^^^^^^^^^
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/43pnqjci11b8rki32vvybms8f1k6v0rw-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     return ctx.invoke(self.callback, **ctx.params)
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/43pnqjci11b8rki32vvybms8f1k6v0rw-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 783, in invoke
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     return __callback(*args, **kwargs)
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/yf5x4z7yir309qd3qyyn8w06kpxcfzxj-python3.11-auto-cpufreq-2.3.0/lib/python3.11/site-packages/auto_cpufreq/bin/auto_cpufreq.py", line 123, in main
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     battery_setup()
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/yf5x4z7yir309qd3qyyn8w06kpxcfzxj-python3.11-auto-cpufreq-2.3.0/lib/python3.11/site-packages/auto_cpufreq/battery_scripts/battery.py", line 17, in battery_setup
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     if lsmod("ideapad_acpi"): ideapad_acpi_setup()
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:        ^^^^^^^^^^^^^^^^^^^^^
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/yf5x4z7yir309qd3qyyn8w06kpxcfzxj-python3.11-auto-cpufreq-2.3.0/lib/python3.11/site-packages/auto_cpufreq/battery_scripts/battery.py", line 8, in lsmod
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     def lsmod(module): return module in run(['lsmod'], stdout=PIPE, stderr=PIPE, text=True).stdout
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/pfv4raslwhi3101k342752v65zxkwrxq-python3-3.11.9/lib/python3.11/subprocess.py", line 548, in run
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     with Popen(*popenargs, **kwargs) as process:
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/pfv4raslwhi3101k342752v65zxkwrxq-python3-3.11.9/lib/python3.11/subprocess.py", line 1026, in __init__
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     self._execute_child(args, executable, preexec_fn, close_fds,
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:   File "/nix/store/pfv4raslwhi3101k342752v65zxkwrxq-python3-3.11.9/lib/python3.11/subprocess.py", line 1955, in _execute_child
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]:     raise child_exception_type(errno_num, err_msg, err_filename)
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]: FileNotFoundError: [Errno 2] No such file or directory: 'lsmod'
Jul 29 11:14:16 perrrkele auto-cpufreq[237961]: Using settings defined in /nix/store/8ircxfaycrw2b8gbrd16bg53bvhmwvlk-auto-cpufreq.conf file
Jul 29 11:14:16 perrrkele systemd[1]: auto-cpufreq.service: Main process exited, code=exited, status=1/FAILURE
```
---

### System information:

Add/paste output of:

```
sudo auto-cpufreq --debug

-------------------------------------------------------------------------------

Linux distro: NixOS 24.05 Uakari
Linux kernel: 6.9.11-xanmod1
Processor: 11th Gen Intel(R) Core(TM) i7-11370H @ 3.30GHz
Cores: 8
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3301 MHz
CPU min frequency: 400 MHz

Core    Usage   Temperature     Frequency
CPU0      0.0%        42 °C      4122 MHz
CPU1      2.0%        43 °C      3301 MHz
CPU2      0.0%        39 °C      3301 MHz
CPU3      1.0%        39 °C      3301 MHz
CPU4      2.0%        42 °C      3138 MHz
CPU5      1.0%        43 °C      3301 MHz
CPU6      0.0%        39 °C      2721 MHz
CPU7      0.0%        39 °C       887 MHz

auto-cpufreq version: 2.3.0

Python: 3.11.9
psutil package: 5.9.8
platform package: 1.0.8
click package: 8.1.7
distro package: 1.9.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.23 %

Total CPU usage: 1.8 %
Total system load: 0.43
Average temp. of all cores: 40.75 °C 

Currently using: performance governor
Currently turbo boost is: on

-------------------------------------------------------------------------------
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

PurpleWazard, 2024-07-29:
lsmod is a command that lists the currently running kernel modules. we use this to check which laptop you have, and use the corresponding paths. type lsmod in your terminal does it give you an error? do you have lsmod installed?

shadeyg56, 2024-07-29:
Duplicate of #740

closed 2024-07-29 by shadeyg56