AdnanHodzic/auto-cpufreq #758: should we add Code Docs for Devs

PurpleWazard, 2024-08-02:
so i had an idea for the devs on this project. since its been a while since ive been able to help with this, some stuff has changed. i thought there should be a markdown file or something that explains each function in brief or causal detail on how it works. and also how the project works as a whole. 

i think this would be beneficial to new devs. because they wouldn't have to figure out for them selves how certain parts work. also it would summarize what does what and how it does it.

idk if this is something yall would want. let me know.

AdnanHodzic, 2024-08-04:
Absolutely, this would be amazing to have. Please go for it!

PurpleWazard, 2024-08-04:
will do.

closed 2024-08-04 by PurpleWazard