AdnanHodzic/auto-cpufreq #763: fan is running and I'm getting "CPU fan speed: 0 RPM"

RayenMnif, 2024-08-12:
### System information:
output of `auto-cpufreq --debug`
```
-------------------------------------------------------------------------------

Linux distro: Arch Linux rolling n/a
Linux kernel: 6.6.44-3-lts
Processor: AMD Ryzen 5 5600H with Radeon Graphics
Cores: 12
Architecture: x86_64
Driver: amd-pstate-epp

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 4280 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0      3.0%        50 °C       400 MHz
CPU1      2.0%        50 °C       400 MHz
CPU2      0.0%        50 °C      3360 MHz
CPU3      0.0%        50 °C      3275 MHz
CPU4      0.0%        50 °C       400 MHz
CPU5      5.8%        50 °C       400 MHz
CPU6      2.0%        50 °C      3291 MHz
CPU7      0.0%        50 °C       400 MHz
CPU8      2.0%        50 °C      3351 MHz
CPU9      0.0%        50 °C      3308 MHz
CPU10      0.0%        50 °C       400 MHz
CPU11      2.9%        50 °C       400 MHz

CPU fan speed: 0 RPM

auto-cpufreq version: Version         : 2.3.0-1

Python: 3.12.4
psutil package: 5.9.8
platform package: 1.0.8
click package: 8.1.7
distro package: 1.9.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.2 %

Total CPU usage: 1.1 %
Total system load: 0.63
Average temp. of all cores: 50.00 °C

Currently using: performance governor
CPU turbo is controlled by amd-pstate-epp driver
Currently turbo boost is: off
```

AdnanHodzic, 2024-08-18:
Can you confirm you never saw it above 0RPM?

Looking at other stats, it truly seems your CPU fan was at 0 RPM, load is low, CPU usage same thing, so thing this is expected behavior.

RayenMnif, 2024-08-20:
@AdnanHodzic yes 
[image]

Heavygrass, 2025-09-25:
Hi,
I'm also **always** observing 0 RPM on Zen 3, 7th series Ryzen.
kernel: 6.14.0-29-generic.
Can hear fan running very well for sure.
Architecture: x86_64   
Driver: amd-pstate-epp

etoimene, 2026-01-17:
I think this might be related to ASUS. Do you have ASUS computer? 
I have
```
Machine:
  Type: Laptop System: ASUSTeK product: Zenbook UM3402YAR_UM3402YA v: 1.0
```
I keep hearing fan(s) in the background. Also, laptop seems quite warm which is OK now in the winter but when warmer days come ... :(

I hope to achieve something by using auto-cpufreq.

With the latest version of auto-cpufreq, fan speed is not even reported.

```
Linux distro: Linux Mint 22.2 Zara
Linux kernel: 6.14.0-37-generic
Processor: AMD Ryzen 7 7730U with Radeon Graphics
Cores: 16
Architecture: x86_64
Driver: amd-pstate-epp

auto-cpufreq version: 3.0.0
```