AdnanHodzic/auto-cpufreq #769: installation failed OpenSuse 15.6

emirhurem, 2024-09-04:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```────────────────────────────────────────── auto-cpufreq installer ──────────────────────────────────────────

Welcome to auto-cpufreq tool installer.

Select a key [I]nstall/[R]emove or press ctrl+c to quit: i

────────────────────────────────────── Detected OpenSUSE distribution ──────────────────────────────────────

─────────────────────────────────────── Setting up Python environment ───────────────────────────────────────

Loading repository data...
Reading installed packages...
'python3-setuptools' is already installed.
No update candidate for 'python3-setuptools-44.1.1-150400.9.9.1.noarch'. The highest available version is already installed.
'dmidecode' is already installed.
No update candidate for 'dmidecode-3.6-150400.16.11.2.x86_64'. The highest available version is already installed.
'python3' is already installed.
No update candidate for 'python3-3.6.15-150300.10.65.2.x86_64'. The highest available version is already installed.
'gcc' is already installed.
No update candidate for 'gcc-7-3.9.1.x86_64'. The highest available version is already installed.
'gtk3' not found in package names. Trying capabilities.
'libgtk-3-0' providing 'gtk3' is already installed.
'python3-pip' is already installed.
No update candidate for 'python3-pip-20.0.2-150400.20.1.noarch'. The highest available version is already installed.
Resolving package dependencies...

The following 2 recommended packages were automatically selected:
  libpng16-compat-devel pcre2-doc

The following 98 NEW packages are going to be installed:
  at-spi2-core at-spi2-core-devel at-spi2-core-lang cairo-devel compositeproto-devel damageproto-devel
  dbus-1-devel fixesproto-devel fontconfig-devel freetype2-devel fribidi-devel gdk-pixbuf-devel
  gettext-its-gtk4 glib2-devel gobject-introspection gobject-introspection-devel graphite2-devel gtk3-devel
  harfbuzz-devel inputproto-devel kbproto-devel libblkid-devel libbrotli-devel libbz2-devel libdatrie-devel
  libepoxy-devel libexpat-devel libffi-devel libglvnd-devel libgthread-2_0-0 libharfbuzz-cairo0
  libharfbuzz-gobject0 libharfbuzz-icu0 libicu73_2-devel libjpeg62 libjpeg62-devel libmount-devel
  libpcre2-32-0 libpcre2-posix3 libpixman-1-0-devel libpng16-compat-devel libpng16-devel libselinux-devel
  libsepol-devel libstdc++6-devel-gcc7 libstdc++-devel libthai-devel libtiff-devel libX11-devel libXau-devel
  libxcb-devel libxcb-screensaver0 libxcb-xf86dri0 libxcb-xtest0 libxcb-xvmc0 libXcomposite-devel
  libXcursor-devel libXdamage-devel libXext-devel libXfixes-devel libXft-devel libXi-devel libXinerama-devel
  libxkbcommon-devel libXrandr-devel libXrender-devel libXtst-devel libzstd-devel Mesa-KHR-devel
  Mesa-libEGL-devel Mesa-libGL-devel pango-devel pcre2-devel pcre2-doc pcre2-tools pthread-stubs-devel
  python3-cairo python3-cairo-devel python3-devel python3-gobject-cairo python3-gobject-Gdk
  python-cairo-common-devel randrproto-devel recordproto-devel renderproto-devel typelib-1_0-Atk-1_0
  typelib-1_0-Atspi-2_0 typelib-1_0-GdkPixbuf-2_0 typelib-1_0-GdkPixdata-2_0 typelib-1_0-Gtk-3_0
  typelib-1_0-HarfBuzz-0_0 typelib-1_0-Pango-1_0 wayland-devel xextproto-devel xineramaproto-devel
  xproto-devel xz-devel zlib-devel

98 new packages to install.
Overall download size: 22.4 MiB. Already cached: 674.2 KiB. After the operation, additional 124.2 MiB will be
used.

Backend:  classic_rpmtrans
Continue? [y/n/v/...? shows all options] (y): y
Retrieving: Mesa-KHR-devel-23.3.4-150600.85.pm.3.x86_64 (Packman)                        (1/98), 177.2 KiB    
Retrieving: Mesa-KHR-devel-23.3.4-150600.85.pm.3.x86_64.rpm ...............................[done (20.0 KiB/s)]
Retrieving: at-spi2-core-2.50.0-150600.1.2.x86_64 (Main Repository)                      (2/98),  81.9 KiB    
Retrieving: at-spi2-core-2.50.0-150600.1.2.x86_64.rpm ..................................................[done]
Retrieving: compositeproto-devel-0.4.2-1.22.x86_64 (Main Repository)                     (3/98),  15.2 KiB    
Retrieving: compositeproto-devel-0.4.2-1.22.x86_64.rpm .................................................[done]
[... 339 lines cut ...]
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:
Operating System: openSUSE Leap 15.6
KDE Plasma Version: 5.27.11
KDE Frameworks Version: 5.115.0
Qt Version: 5.15.12
Kernel Version: 6.4.0-150600.23.17-default (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i5-6300U CPU @ 2.40GHz
Memory: 15.4 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 520
Manufacturer: Dell Inc.
Product Name: Latitude 7280
```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

AdnanHodzic, 2024-09-05:
It seems like something when wrong when it was trying to pull the poetry-core & poetry-dynamic-versioning package, can you try again or getting it from different repo?

traptoo, 2025-02-16:
Same OS, similar error here:

`────────────────────────────────────────────── Installing auto-cpufreq tool ──────────────────────────────────────────────

Processing /home/phillip/auto-cpufreq
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/auto-cpufreq/venv/bin/python /opt/auto-cpufreq/venv/lib64/python3.6/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-91v62dew/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'poetry-core>=1.0.0' 'poetry-dynamic-versioning>=1.0.0,<2.0.0'
       cwd: None
  Complete output (4 lines):
  Collecting poetry-core>=1.0.0
    Using cached poetry_core-1.0.8-py2.py3-none-any.whl (425 kB)
  ERROR: Could not find a version that satisfies the requirement poetry-dynamic-versioning<2.0.0,>=1.0.0 (from versions: 0.1.0, 0.2.0, 0.3.0, 0.3.1, 0.3.2, 0.4.0, 0.5.0, 0.6.0, 0.7.0, 0.8.0, 0.8.1, 0.8.2, 0.8.3, 0.9.0, 0.10.0, 0.11.0, 0.12.0, 0.12.1, 0.12.2, 0.12.3, 0.12.4, 0.12.5, 0.12.6, 0.12.7, 0.13.0, 0.13.1, 0.14.0, 0.14.1, 0.15.0, 0.16.0, 0.17.0, 0.17.1, 1.0.0b1, 1.0.0b2, 1.0.0b3)
  ERROR: No matching distribution found for poetry-dynamic-versioning<2.0.0,>=1.0.0
  ----------------------------------------
WARNING: Discarding file:///home/phillip/auto-cpufreq. Command errored out with exit status 1: /opt/auto-cpufreq/venv/bin/python /opt/auto-cpufreq/venv/lib64/python3.6/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-91v62dew/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'poetry-core>=1.0.0' 'poetry-dynamic-versioning>=1.0.0,<2.0.0' Check the logs for full command output.
ERROR: Command errored out with exit status 1: /opt/auto-cpufreq/venv/bin/python /opt/auto-cpufreq/venv/lib64/python3.6/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-91v62dew/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'poetry-core>=1.0.0' 'poetry-dynamic-versioning>=1.0.0,<2.0.0' Check the logs for full command output.

──────────────────────────────────────── auto-cpufreq tool successfully installed ────────────────────────────────────────

For list of options, run:
auto-cpufreq --help

`

How to change repo?

AdnanHodzic, 2025-02-17:
Think this is related to #808 

@CaffineAddic if you're looking for way to contribute more, helping with this one would be great 🙂

CaffineAddic, 2025-02-18:
> Think this is related to [#808](https://github.com/AdnanHodzic/auto-cpufreq/pull/808)
> 
> [@CaffineAddic](https://github.com/CaffineAddic) if you're looking for way to contribute more, helping with this one would be great 🙂

Copy that chief, I will create a leap VM and try diagnosing it

CaffineAddic, 2025-02-18:
@AdnanHodzic which file has the python env install script, i.e. the line which is doing the pip install command 

with list of packages.

AdnanHodzic, 2025-02-21:
Not sure if you caught it, but you can also use [auto-cpufreq genAI chatbot](https://foolcontrol.org/?p=4903) for this purpose 😉 

- [pyproject.toml ](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/pyproject.toml)defines the project dependencies but does not lock them to specific versions.
- [poetry.lock](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/poetry.lock) contains the exact versions of all dependencies that Poetry has resolved.

CaffineAddic, 2025-02-21:
@AdnanHodzic Thanks captain, [auto-cpufreq genAI chatbot](https://foolcontrol.org/?p=4903) I used this but it gave subpar responses which were not actually relevant. 😭 

I will look into it Captain, if you are working something open source in A.I., I would love to help there as well

CaffineAddic, 2025-02-21:
@AdnanHodzic  

Captain, I think I found a place of issue 

Here is available versions of 

poetry-dynamic-versioning (1.7.1)

Available versions: 1.7.1, 1.7.0, 1.6.0, 1.5.2, 1.5.1, 1.5.0, 1.4.1, 1.4.0, 1.3.0, 1.2.0, 1.1.1, 1.1.0, 1.0.1, 1.0.0, 0.25.0, 0.24.0, 0.23.0, 0.22.0, 0.21.5, 0.21.4, 0.21.3, 0.21.2, 0.21.1, 0.21.0, 0.20.0, 0.19.0, 0.18.0, 0.17.1, 0.17.0, 0.16.0, 0.15.0, 0.14.1, 0.14.0, 0.13.1, 0.13.0, 0.12.7, 0.12.6, 0.12.5, 0.12.4, 0.12.3, 0.12.2, 0.12.1, 0.12.0, 0.11.0, 0.10.0, 0.9.0, 0.8.3, 0.8.2, 0.8.1, 0.8.0, 0.7.0, 0.6.0, 0.5.0, 0.4.0, 0.3.2, 0.3.1, 0.3.0, 0.2.0, 0.1.0
  
In OpenSUSE Tumbleweed (rolling release)

But in Opensuse Leap the packages are much older and in my experience down right ancient 

ERROR: Could not find a version that satisfies the requirement poetry-dynamic-versioning<2.0.0,>=1.0.0 (from versions: 0.1.0, 0.2.0, 0.3.0, 0.3.1, 0.3.2, 0.4.0, 0.5.0, 0.6.0, 0.7.0, 0.8.0, 0.8.1, 0.8.2, 0.8.3, 0.9.0, 0.10.0, 0.11.0, 0.12.0, 0.12.1, 0.12.2, 0.12.3, 0.12.4, 0.12.5, 0.12.6, 0.12.7, 0.13.0, 0.13.1, 0.14.0, 0.14.1, 0.15.0, 0.16.0, 0.17.0, 0.17.1, 1.0.0b1, 1.0.0b2, 1.0.0b3)
  ERROR: No matching distribution found for poetry-dynamic-versioning<2.0.0,>=1.0.0
 

To solve >= 1.0.0, leap doesn't have those latest packages are it's using Pip 20.0.2

I think there are two ways to solve it

 poetry-dynamic-versioning<2.0.0,>=1.0.0-beta can be added to the pyproject.toml to allow beta versions for all 
or add an exception for leap.

AdnanHodzic, 2025-02-23:
> poetry-dynamic-versioning<2.0.0,>=1.0.0-beta can be added to the pyproject.toml to allow beta versions for all
or add an exception for leap.

If you could please create a pull request for this, let's see if it solves it for @emirhurem and I can also test if it breaks anything else along the way.

CaffineAddic, 2025-02-23:
Copy that captain, on it.

CaffineAddic, 2025-02-23:
Dynamic versioning plugin for Poetry

This is a Python 3.7+ plugin for [Poetry 1.2.0+](https://github.com/sdispater/poetry) and [Poetry Core 1.0.0+](https://github.com/python-poetry/poetry-core)

and the default install that OpenSuse Leap does is, 

It goes for poetry 1.1.15, if allowed pip install poetry.

I don't know how the env is created in auto-cpufreq because of this versioning nothing works - even letting the Beta versions to be available.

CaffineAddic, 2025-03-01:
@AdnanHodzic Captain, I haven't created a pull request as doing those changes still don't allow the auto-cpufreq to work

AdnanHodzic, 2025-03-01:
@CaffineAddic thanks for the update, no worries, take your time!

CaffineAddic, 2025-03-23:
@traptoo Did you try the snap install?