AdnanHodzic/auto-cpufreq #796: Facing severe performance issue: All CPU at minimum CPU frequency in Performance mode

lamyergeier, 2024-11-26:
### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted?  Yes
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? No
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? No

---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

```

Linux distro: Fedora Linux 41 Workstation Edition
Linux kernel: 6.11.8-300.fc41.x86_64
Processor: 12th Gen Intel(R) Core(TM) i5-1240P
Cores: 16
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 4400 MHz
CPU min frequency: 400 MHz

Core    Usage   Temperature     Frequency
CPU0     62.2%        60 °C       400 MHz
CPU1     81.0%        60 °C       400 MHz
CPU2     54.6%        59 °C       400 MHz
CPU3     95.0%        59 °C       400 MHz
CPU4     83.2%        59 °C       400 MHz
CPU5     80.8%        59 °C       400 MHz
CPU6     81.8%        58 °C       400 MHz
CPU7     77.8%        58 °C       400 MHz
CPU8     91.9%        59 °C       400 MHz
CPU9     91.9%        59 °C       400 MHz
CPU10     91.9%        59 °C       400 MHz
CPU11    100.0%        59 °C       400 MHz
CPU12     99.0%        60 °C       400 MHz
CPU13    100.0%        60 °C       400 MHz
CPU14    100.0%        59 °C       400 MHz
CPU15    100.0%        60 °C       400 MHz

auto-cpufreq version: 2.4.0

Python: 3.13.0
psutil package: 6.1.0
platform package: 1.0.8
click package: 8.1.7
distro package: 1.9.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.1 %

Total CPU usage: 99.0 %
Total system load: 14.79
Average temp. of all cores: 59.19 °C 

Currently using: performance governor
Currently turbo boost is: on
```

[image]

---

AdnanHodzic, 2024-11-29:
Is it any different if you [change the driver](https://github.com/AdnanHodzic/auto-cpufreq?tab=readme-ov-file#troubleshooting) or use a [config file](https://github.com/AdnanHodzic/auto-cpufreq?tab=readme-ov-file#3-auto-cpufreq-config-file)?

ryuheechul, 2025-01-05:
I'm also experiencing the same thing.

(`performance` governor automatically kicks in when there is AC power connected and `schedutil` and way slower with performance governor ironically due to the frequency limited to 400 MHz - reported via `cpupower frequency-info` as "current CPU frequency". But `performance` governor set as override - while not connected to power - would not suffer the same issue mysteriously)

It's happening across multiple different kernel versions ranging at least from 6.6.68 to 6.12.7.

And drivers both with 
`amd-pstate-epp` and `acpi-cpufreq`.

The device experiencing this is GPD Win Max 2 (AMD) on NixOS

ryuheechul, 2025-01-06:
Follow up on my comment above, I'm experiencing this issue even without `auto-cpufreq` being used and now I see some search result that people report their Linux machine slow down quite a lot on charging (with some low battery level, probably low enough to cause actual charging to kick off and people speculate that it might have to do with heat caused by charging). In fact, when my device's battery is full and connected to charger, this issue kind of disappears which seems to back the stories that people share with their experience.