AdnanHodzic/auto-cpufreq #800: Not sure it works with intel_cpupref

eugenialoli, 2024-12-14:
I set up the file on /etc as such

[battery]
governor = powersave
turbo = never
scaling_max_freq = 1000000

but it still goes to 1.5 Ghz, it doesn't stop on 1 ghz. In fact, most of the time, even on idling cpus, it goes to 1.5 ghz. Very rarely it goes to 500 mhz.

sudo auto-cpufreq --debug

Using settings defined in /etc/auto-cpufreq.conf file

-------------------------------------------------------------------------------

Linux distro: Linux Mint 22 Wilma
Linux kernel: 6.8.0-50-generic
Processor: Intel(R) Core(TM) i5-5250U CPU @ 1.60GHz
Cores: 4
Architecture: x86_64
Driver: intel_cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1600 MHz
CPU min frequency: 500 MHz

Core	Usage	Temperature	Frequency
CPU0      4.0%        47 °C      1500 MHz
CPU1      5.0%        48 °C      1500 MHz
CPU2      3.0%        47 °C      1500 MHz
CPU3      2.0%        48 °C      1500 MHz

CPU fan speed: 1186 RPM

auto-cpufreq version: 2.4.0

Python: 3.12.3
psutil package: 6.1.0
platform package: 1.0.8
click package: 8.1.7
distro package: 1.9.0

Computer type: Laptop
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.39 %

Total CPU usage: 1.0 %
Total system load: 0.49
Average temp. of all cores: 47.50 °C 

Currently using: powersave governor
Currently turbo boost is: off

[image]

kdgnlx, 2024-12-21:
I am experiencing the same issue. I edited the file but cpu frequency goes above the scaling_max_freq.
It's also very odd the high cpu temperatures I see in cinnamon compared to other DEs like gnome or kde.

AdnanHodzic, 2025-04-02:
There was an old issue regarding this, that after making `scaling_max_freq` or `scaling_min_freq` changes will not be automatically applied and that easiest way to get it working is to restart your computer or auto-cpufreq systemd service.

Can you report if the problem persists after you done this?

AdnanHodzic, 2025-07-20:
Changes in #866 are merged with `master` which should fix this issue, please pull the latest changes.