AdnanHodzic/auto-cpufreq #812: How to configure permissions to not require pswd for Governor/power mode change?

Heavygrass, 2025-03-06:
Hi!

Is it possible to set file permissions in Linux (or do anything else) to **avoid** auto-cpufreq **requiring password** for each **Governor/power mode change**?
Its currently asking for (root permissions? I guess) every time I switch Governor/power mode (at least on Cinnamon).
If this is possible, which files one need to change ownership or permissions?

I did a search of issues and couldn't find anything relevant, please let me know if there's.

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
Yes.

Thank you for this great project!

shadeyg56, 2025-03-06:
The files `auto-cpufreq` edits are in sysfs which is provided by the kernel and they come from the `cpufreq` driver. It is up to the driver to set the owner of the attributes it exposes. You might be able to make a group and combine that with some udev rules or something, but I wouldn't recommend messing with it too much since it could lead to unexpected behavior.

closed 2025-03-06 by shadeyg56

AdnanHodzic, 2025-03-06:
@Heavygrass you can simply configure [sudo NOPASSWD in sudoers file](https://www.tecmint.com/run-sudo-command-without-password-linux/) not to be asked for password every time you want to run something as root.

Heavygrass, 2025-03-06:
@AdnanHodzic Thank you for your response!
Is removing root password the only way?

AdnanHodzic, 2025-03-06:
Who said anything about removing root password? 🤔

Article I shared above instructs you how to not be prompted for password every time you want to run something as root.

Heavygrass, 2025-03-14:
@shadeyg56 @AdnanHodzic 
Is it possible to point me to say a discussion (or anything else) that could help me to understand the reason(s) for requiring sudo for governor change design decision was made?
Thank you.

AdnanHodzic, 2025-03-15:
You're asking questions that don't have anything to do with auto-cpufreq and are related to how Linux itself and some of its operations were designed.

If you want to change e.g CPU governor state in Linux, you need sudo/root permissions to perform this action (you can see this in code btw). Hence if you want auto-cpufreq to automatically make these decisions for you, you need to provide it with root access, simple as that.

I can't find exact discussions now (which are somewhere in issues), but that's why auto-cpufreq won't ask you for root for actions where it's not needed. For example, you can run `auto-cpufreq --stats` (without sudo) as it's not making any actions and is just printing information.

ekasprzak, 2025-11-16:
Hi @AdnanHodzic, do you know a way to disable asking for password ONLY when using auto-cpufreq GUI?
I added /usr/local/bin/auto-cpufreq to sudoers and can change governor using CLI without a password, but GUI still asks for it.

developpeur-php, 2025-11-19:
Using pkexec in the GUI code appears to be the source of the problem, but I'm not making any definitive claims.

https://man.archlinux.org/man/pkexec.1
> Executing a program as another user is a privileged operation. By default the action to check for (see the section called “ACTION AND AUTHORIZATIONS”) requires administrator authentication. In addition, the authentication dialog presented to the user will display the full path to the program to be executed so the user is aware of what will happen.

Heavygrass, 2025-11-20:
How/what modifies `/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor`?
And `/sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference`?

Heavygrass, 2025-11-20:
> [@ekasprzak](https://github.com/ekasprzak) [@Heavygrass](https://github.com/Heavygrass) Besides this ^ can you please pull the latest changes

I've never built auto-cpufreq yet sorry (simply use automatic update). If there's a build for Ubuntu Cinnamon 24.04 I'm happy to test it.

developpeur-php, 2025-11-20:
> How/what modifies `/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor`? And `/sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference`?

A file named _cpufreqctl.sh_ in the project is installed as _cpufreqctl.auto-cpufreq_ in _/usr/local/bin_.
_auto-cpufreq_ use _cpufreqctl.auto-cpufreq_ to modify _/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor_ and _/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor_, make no sense to me 🤔.
But why are you asking?

> > [@ekasprzak](https://github.com/ekasprzak) [@Heavygrass](https://github.com/Heavygrass) Besides this ^ can you please pull the latest changes
> 
> I've never built auto-cpufreq yet sorry (simply use automatic update). If there's a build for Ubuntu Cinnamon 24.04 I'm happy to test it.

Just download the source ZIP from GitHub, go to the project folder, run `sudo ./auto-cpufreq-installer`, and type i to install or r to remove it's that simple 🙂.