AdnanHodzic/auto-cpufreq #817: psutil 7.0.0

LordGrimmauld, 2025-03-18:
psutil version 7.0.0 was released [upstream](https://pypi.org/project/psutil/#history). I saw this project requests any 6.x version. The [release notes](https://github.com/giampaolo/psutil/blob/master/HISTORY.rst#700) only list incompatibilities with python2.7 (this project already uses at minimum 3.7), and removal of `memory_info_ex` which this project does not use at all. This means 7.0.0 should "just work". Any chance to unpin this in the `pyproject.toml`?

Currently, attempting to build against 7.0.0 obviously fails (which i noticed in the [nixpkgs ci](https://hydra.nixos.org/build/292453144/log)). Figured i'd open an issue, partially to track for myself. I understand this is primarily a packaging thing and failing on the nix side, but by the looks of it supporting the new release of psutil shouldn't be too hard.

AdnanHodzic, 2025-03-22:
Similar thing was discussed in #813 in regards to `girepository-2.0`. 

Regarding unpinning `psutil`, I'm completely fine with this, I encourage you to give it a try yourself with changes you would like to see and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

LordGrimmauld, 2025-03-22:
Alright, i opened https://github.com/AdnanHodzic/auto-cpufreq/pull/821

PR #821 (2025-03-22): deps: allow psutil 7.x
This allows building with both psutil 6.x and 7.x. Closes #817.

Upstream [1] released psutil 7.0.0 recently, which caused a build failure on nixpkgs ci [2].
Checking release notes [3], `memory_info_ex` was removed (unused in auto-cpufreq),
and minimum python version requirements were increased to what this project uses already anyways.

[1] https://pypi.org/project/psutil/#history
[2] https://hydra.nixos.org/build/292453144/log
[3] https://github.com/giampaolo/psutil/blob/master/HISTORY.rst#700

Tested with:
```patch
diff --git a/flake.nix b/flake.nix
index ab3449b..89a [... cut]
commit 8f026ac6 (2025-03-22): deps: allow psutil 7.x (#821)

This allows building with both psutil 6.x and 7.x. Closes #817.

Upstream [1] released psutil 7.0.0 recently, which caused a build failure on nixpkgs ci [2].
Checking release notes [3], `memory_info_ex` was removed (unused in auto-cpufreq),
and minimum python version requirements were increased to what this project uses already anyways.

[1] https://pypi.org/project/psutil/#history
[2] https://hydra.nixos.org/build/292453144/log
[3] https://github.com/giampaolo/psutil/blob/master/HISTORY.rst#700

Tested with:
```patch
diff --git a/flake.nix b/flake.nix
index ab34 [... cut]
closed 2025-03-22 by AdnanHodzic