AdnanHodzic/auto-cpufreq #818: Warning: Changing CPU turbo is not supported. Skipping.

brook1167, 2025-03-18:
I disabled Turbo Boost in the BIOS settings, but I am still getting the following warning when running the application:

LAITH343, 2025-03-20:
this is normal behavior because it doesn't know your turbo boost is disabled

brook1167, 2025-03-21:
If it's the cause, how can I know that auth-cpu-freq is working?

AdnanHodzic, 2025-03-22:
Fact you're seeing those stats above means that it's working. 

Only thing that isn't working is setting your CPU to use turbo because you're hardware doesn't support it. Other then that governors are changed and it looks good to me.

For more info I advise you to check out [auto-cpufreq - tool demo (Automatic CPU speed & power optimizer for Linux)](https://youtu.be/QkYRpVEEIlg) or [auto-cpufreq v2.0 release & demo of all available features and options](https://youtu.be/SPGpkZ0AZVU) Youtube videos.

closed 2025-03-22 by AdnanHodzic