AdnanHodzic/auto-cpufreq #825: GUI not launching on linux mint Cinnamon edition

kirmola, 2025-03-28:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

#### Before submitting an issue, it is strongly recommended to use the **[auto-cpufreq-genAI-chatbot](https://foolcontrol.org/?p=4903)** to get an immediate answer to your question.

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
 ### YES, one issue related to this was closed for not filling this template
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
### Not Applicable
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
### Yes, N/A
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?
### N/A

### Error output:
```text
kirmola@kirmola-IdeaPad-5-15ITL05:~$ sudo auto-cpufreq-gtk 
Traceback (most recent call last):
  File "/opt/auto-cpufreq/venv/bin/auto-cpufreq-gtk", line 5, in <module>
    from auto_cpufreq.bin.auto_cpufreq_gtk import main
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/bin/auto_cpufreq_gtk.py", line 2, in <module>
    import gi
ModuleNotFoundError: No module named 'gi'

```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```
```

-------------------------------- Battery Info ---------------------------------

battery count = 1
cat: /sys/class/power_supply/BAT0/charge_start_threshold: No such file or directory
ERROR: failed to read battery BAT0 thresholds: CalledProcessError(1, ['cat', '/sys/class/power_supply/BAT0/charge_start_threshold'])

-------------------------------------------------------------------------------

Linux distro: Linux Mint 22.1 Xia
Linux kernel: 6.8.0-56-generic
Processor: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2400 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0      5.2%        35 °C       400 MHz
CPU1      2.0%        36 °C      1244 MHz
CPU2      2.0%        34 °C       400 MHz
CPU3      2.0%        32 °C      1174 MHz
CPU4      3.3%        35 °C      1241 MHz
CPU5      3.1%        36 °C      1216 MHz
CPU6      2.0%        34 °C      1169 MHz
CPU7      3.0%        32 °C       779 MHz

auto-cpufreq version: 2.5.0

Python: 3.12.3
psutil package: 7.0.0
platform package: 1.0.8
click package: 8.1.8
distro package: 1.9.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.23 %

Total CPU usage: 1.3 %
Total system load: 0.29
Average temp. of all cores: 34.00 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---
## GUI should launch but  not launching, I have tried installing 
```
sudo apt install python3-gi
```
as well as from 
```
pip install PyGObject
```
but no success..

kirmola, 2025-03-28:
I found a fix, it seems that  the error from ```pip install PyGObject``` which is required to for the file ```auto_cpufreq_gtk.py``` require ```libgirepository-2.0``` which is available by running ```＄ sudo apt-get install libgirepository-2.0-dev```, I installedit and now its running perfectly, I have to install PyGObject in venv created by auto-cpufreq though. And now GUI is working:

[image]. anyone stumbling on the issue should be able to fix it themselves

AdnanHodzic, 2025-03-28:
Problem with older dependencies was also mentioned in #813

kirmola, 2025-03-28:
all right then, its known then, I'll close this

closed 2025-03-28 by kirmola

AdnanHodzic, 2025-03-28:
> all right then, its known then, I'll close this

No worries, once #826 is merged, I'll set that it closes all these 3 issues :)

AdnanHodzic, 2025-03-30:
With #826 being merged with `master` branch, your problem should be solved. Please re-run auto-cpufreq-installer and auto-cpufreq daemon with latest code changes.