AdnanHodzic/auto-cpufreq #828: Battery Charging Thresholds Not Applied by auto-cpufreq

rakshitx1, 2025-04-02:
# Battery Charging Thresholds Not Applied by auto-cpufreq

## Have you tried?
- ✅ Searched through existing/closed issues.
- ✅ Verified installation via `auto-cpufreq --install` and `auto-cpufreq --remove`.
- ✅ Checked GNOME Power Profiles Daemon status.
- ✅ Reviewed `auto-cpufreq --debug` output.

## Error output:
No explicit error message, but **battery charging thresholds** do not seem to be enforced. 

[image]

## System Information:
```
Linux distro: Ubuntu 22.04 Jammy Jellyfish  
Kernel: 6.8.0-57-generic  
CPU: 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz  
Architecture: x86_64  
Driver: intel_pstate  
auto-cpufreq version: 2.5.0  
```

## Issue Description:
Despite setting charging thresholds in `/etc/auto-cpufreq.conf`:
```
enable_thresholds = true
start_threshold = 60
stop_threshold = 80
```
Auto-cpufreq does not enforce these limits. The battery is charging beyond the stop threshold with **no rules detected** in `auto-cpufreq --stats`. 

## Steps Taken:
- Installed and removed auto-cpufreq to reset settings.
- Verified GNOME Power Profiles Daemon interference.
- Ensured `/etc/auto-cpufreq.conf` is properly configured.
- Observed battery stats showing **no charging thresholds applied**.

## Expected Behavior:
Battery charging should **stop at 80%** and **resume at 60%**, but this is not happening.

Would appreciate any guidance on troubleshooting or confirming whether this is a known bug.

AdnanHodzic, 2025-04-02:
@PurpleWazard 👆

PurpleWazard, 2025-04-02:
Hey @rakshitx1 what is your laptop? Vendor and model. Depending on your vendor it may or may not support battery charge thresholds

rakshitx1, 2025-04-02:
@PurpleWazard 

My laptop is the MSI GF63 11UC (i7-11800H).

I checked `/sys/class/power_supply/BAT1/`, and I see the following charge control files:  

```bash
charge_control_end_threshold
charge_control_start_threshold
```

Let me know if you need any other details!

PurpleWazard, 2025-04-03:
@rakshitx1 alrigt thanks what is the output of lsmod?

rakshitx1, 2025-04-14:
This is output of `lsmod`

```bash
$ lsmod
Module                  Size  Used by
btrfs                2031616  0
blake2b_generic        24576  0
xor                    20480  1 btrfs
raid6_pq              126976  1 btrfs
ufs                   126976  0
qnx4                   12288  0
hfsplus               139264  0
hfs                    77824  0
minix                  49152  0
ntfs                  139264  0
msdos                  16384  0
jfs                   303104  0
nls_ucs2_utils          8192  1 jfs
xfs                  2367488  0
nvidia_uvm           2088960  0
rfcomm                102400  4
xt_set                 20480  0
ip_set                 61440  1 xt_set
xt_addrtype            12288  4
xfrm_user              61440  1
xfrm_algo              16384  1 xfrm_user
ccm                    20480  6
xt_CHECKSUM            12288  1
xt_MASQUERADE          16384  5
xt_conntrack           12288  3
ipt_REJECT             12288  2
nf_reject_ipv4         12288  1 ipt_REJECT
xt_tcpudp              16384  9
nft_compat             20480  24
nft_chain_nat          12288  6
nf_nat                 61440  2 nft_chain_nat,xt_MASQUERADE
nf_conntrack          196608  3 xt_conntrack,nf_nat,xt_MASQUERADE
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         12288  1 nf_conntrack
nf_tables             372736  248 nft_compat,nft_chain_nat
libcrc32c              12288  5 nf_conntrack,nf_nat,btrfs,nf_tables,xfs
nfnetlink              20480  4 nft_compat,nf_tables,ip_set
bridge                421888  0
stp                    12288  1 bridge
llc                    16384  2 bridge,stp
cmac                   12288  2
algif_hash             12288  1
algif_skcipher         16384  1
af_alg                 32768  6 algif_hash,algif_skcipher
overlay               212992  0
bnep                   32768  2
binfmt_misc            24576  1
snd_ctl_led            24576  0
ledtrig_audio          12288  1 snd_ctl_led
snd_soc_skl_hda_dsp    24576  6
snd_soc_hdac_hdmi      45056  1 snd_soc_skl_hda_dsp
snd_sof_probes         20480  0
snd_soc_intel_hda_dsp_common    16384  1 snd_soc_skl_hda_dsp
snd_hda_codec_hdmi     94208  1
snd_hda_codec_realtek   200704  1
snd_soc_dmic           12288  1
xe                   2719744  0
snd_hda_codec_generic   122880  1 snd_hda_codec_realtek
intel_rapl_msr         20480  0
intel_rapl_common      40960  1 intel_rapl_msr
nvidia_drm            131072  2
intel_uncore_frequency    16384  0
snd_sof_pci_intel_tgl    12288  0
intel_uncore_frequency_common    16384  1 intel_uncore_frequency
nvidia_modeset       1548288  2 nvidia_drm
[... 136 lines cut ...]
wmi                    28672  3 video,wmi_bmof,msi_wmi
pinctrl_tigerlake      28672  3
```

In the meantime, I’ve created a systemd service to set battery charge thresholds:

```bash
# /etc/systemd/system/battery-threshold.service
[Unit]
Description=Set Battery Charge Start and Stop Threshold
After=multi-user.target

[Service]
Type=oneshot
ExecStart=/bin/sh -c 'echo 60 > /sys/class/power_supply/BAT1/charge_control_start_threshold && echo 80 > /sys/class/power_supply/BAT1/charge_control_end_threshold'
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```

This works for now, but I also noticed an issue when exiting `auto-cpufreq --live`:
`Failed to start power-profiles-daemon.service: Unit power-profiles-daemon.service is masked.`

Let me know if anything from the lsmod output stands out or if you'd like me to test anything else.

neut, 2025-06-06:
I have the same issue on a Lenovo T14s Gen 1

fiskhest, 2025-07-26:
Also noticing the same issue on a `Dell Precision 5480`. The custom service (tweaking `BAT1` -> `BAT0`) seems to work when I manually verify by `cat`ing `/sys/class/...` files,, but `auto-cpufreq --stats` still reports None for the thresholds.