AdnanHodzic/auto-cpufreq #833: Pop OS install fails

noncarbonatedclack, 2025-04-08:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

#### Before submitting an issue, it is strongly recommended to use the **[auto-cpufreq-genAI-chatbot](https://foolcontrol.org/?p=4903)** to get an immediate answer to your question.

The chatbot provided a link that gave me an error 403 forbidden.

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? - Yes, there are similar issues
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? - No, i do not use snaps
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? - No, install fails
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? - No, install fails

### Error output:
```
> sudo auto-cpufreq/auto-cpufreq-installer 

───────────────────────────────────────────────────────────────────────── auto-cpufreq installer ─────────────────────────────────────────────────────────────────────────

Welcome to auto-cpufreq tool installer.

Select a key [I]nstall/[R]emove or press ctrl+c to quit: i

libgirepository-2.0-dev needs to be installed for version bookworm/sid
---- 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libgirepository-2.0-dev
E: Couldn't find any package by glob 'libgirepository-2.0-dev'

────────────────────────────────────────────────────────────────── Installing necessary Python packages ──────────────────────────────────────────────────────────────────

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (25.0.1)
Requirement already satisfied: wheel in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (0.45.1)
Collecting PyGObject
  Using cached pygobject-3.52.3.tar.gz (1.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [24 lines of output]
      + meson setup /tmp/pip-install-wct09o9d/pygobject_07b7bbccae574a72a3193b6a33623d35 /tmp/pip-install-wct09o9d/pygobject_07b7bbccae574a72a3193b6a33623d35/.mesonpy-_mdmf556 -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md -Dtests=false -Dwheel=true --wrap-mode=nofallback --native-file=/tmp/pip-install-wct09o9d/pygobject_07b7bbccae574a72a3193b6a33623d35/.mesonpy-_mdmf556/meson-python-native-file.ini
      The Meson build system
      Version: 1.7.2
      Source dir: /tmp/pip-install-wct09o9d/pygobject_07b7bbccae574a72a3193b6a33623d35
      Build dir: /tmp/pip-install-wct09o9d/pygobject_07b7bbccae574a72a3193b6a33623d35/.mesonpy-_mdmf556
      Build type: native build
      Project name: pygobject
      Project version: 3.52.3
      C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
      C linker for the host machine: cc ld.bfd 2.38
      Host machine cpu family: x86_64
      Host machine cpu: x86_64
      Program python3 found: YES (/opt/auto-cpufreq/venv/bin/python3)
      Found pkg-config: YES (/usr/bin/pkg-config) 0.29.2
      Run-time dependency python found: YES 3.10
      Did not find CMake 'cmake'
      Found CMake: NO
      Run-time dependency girepository-2.0 found: NO (tried pkgconfig and cmake)
      Not looking for a fallback subproject for the dependency girepository-2.0 because:
      Use of fallback dependencies is disabled.
      
      ../meson.build:31:9: ERROR: Dependency 'girepository-2.0' is required but not found.
      
      A full log can be found at /tmp/pip-install-wct09o9d/pygobject_07b7bbccae574a72a3193b6a33623d35/.mesonpy-_mdmf556/meson-logs/meson-log.txt
[... 89 lines cut ...]
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.22 %

Total CPU usage: 1.0 %
Total system load: 0.70
Average temp. of all cores: 37.00 °C 

Currently using: powersave governor
Currently turbo boost is: on

```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

Expected behavior is a successful install. The installer script states it completed successfully, however, I'm unable to open auto-cpu-freq.

I've tried installing just the missing package but that is also failing..
I've tried locating instructions on enabling fallback dependencies

It's worth noting that I had this working on the same laptop, on Pop OS, a few months ago. I reinstalled Pop a few days ago and am now unable to install auto-cpu-freq, not sure what has changed.

---

AdnanHodzic, 2025-04-08:
Related to improvements made as part of #826. 

@kirmola in this case it's trying to install `libgirepository-2.0-dev` when it shouldn't as it won't find it for 22.04 repos.

AdnanHodzic, 2025-04-09:
@kirmola we have another issue (#834) as outcome of #826, can you please create a PR to address these? Thanks

AdnanHodzic, 2025-04-10:
@noncarbonatedclack & @noamtu123 since @kirmola isn't responding, I pushed a fix as part of [8870b62 commit](https://github.com/AdnanHodzic/auto-cpufreq/commit/8870b62b172b11fea9de63971158a76c271d939a) with revamped logic to solve this issue.

Can you please pull the latest git changes and run `sudo ./auto-cpufreq-installer --install` again.

noamtu123, 2025-04-10:
thank for your quick response i don't think it's working i have tried even completely removing auto-cpufreq then reinstalling and it didn't help

_this what i did andpc info: dell inspiron 14 5459 (intel i5-6200u, amd radeon r5 335 2gb ddr3 vram, 8gb ddr3l ram walram 256gb sata ssd from aliexpress)
Linux distro: Zorin OS 17.3 lite

that was my install process in terminal: what is said:_

```
noamtu123@noamtu123-Inspiron-5459:~$ git clone https://github.com/AdnanHodzic/auto-cpufreq.git
cd auto-cpufreq && sudo ./auto-cpufreq-installer
Cloning into 'auto-cpufreq'...
remote: Enumerating objects: 2837, done.
remote: Counting objects: 100% (906/906), done.
remote: Compressing objects: 100% (246/246), done.
remote: Total 2837 (delta 786), reused 661 (delta 660), pack-reused 1931 (from 2)
Receiving objects: 100% (2837/2837), 961.02 KiB | 2.55 MiB/s, done.
Resolving deltas: 100% (1585/1585), done.
[sudo] password for noamtu123:      

─────────────────────────────────────────────────────────────────────── auto-cpufreq installer ───────────────────────────────────────────────────────────────────────

Welcome to auto-cpufreq tool installer.

Select a key [I]nstall/[R]emove or press ctrl+c to quit: i

───────────────────────────────────────────────────────────────── Detected Debian based distribution ─────────────────────────────────────────────────────────────────

──────────────────────────────────────────────────────────────────── Setting up Python environment ────────────────────────────────────────────────────────────────────

Hit:1 http://il.archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://il.archive.ubuntu.com/ubuntu jammy-updates InRelease                                                                              
Hit:3 http://security.ubuntu.com/ubuntu jammy-security InRelease                                                                               
Hit:4 http://il.archive.ubuntu.com/ubuntu jammy-backports InRelease                                                                                         
Hit:5 https://packages.mozilla.org/apt mozilla InRelease                                                                                                    
Hit:6 https://ppa.launchpadcontent.net/zorinos/apps/ubuntu jammy InRelease                        
Hit:7 https://ppa.launchpadcontent.net/zorinos/drivers/ubuntu jammy InRelease
Hit:8 https://ppa.launchpadcontent.net/zorinos/patches/ubuntu jammy InRelease
Hit:9 https://ppa.launchpadcontent.net/zorinos/stable/ubuntu jammy InRelease
Hit:10 https://packages.zorinos.com/stable jammy InRelease
Hit:11 https://packages.zorinos.com/patches jammy InRelease
Hit:12 https://packages.zorinos.com/apps jammy InRelease
Hit:13 https://packages.zorinos.com/drivers jammy InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
Installing libgirepository1.0-dev...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libgirepository1.0-dev is already the newest version (1.72.0-1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
needs to be installed for version bookworm/sid
---- 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gcc is already the newest version (4:11.2.0-1ubuntu1).
libcairo2-dev is already the newest version (1.16.0-5ubuntu2).
dmidecode is already the newest version (3.3-3ubuntu0.2).
libgtk-3-dev is already the newest version (3.24.33-1ubuntu2.2).
python3-dev is already the newest version (3.10.6-1~22.04.1).
python3-setuptools is already the newest version (59.6.0-1.2ubuntu0.22.04.2).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.5).
python3-venv is already the newest version (3.10.6-1~22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

──────────────────────────────────────────────────────────────── Installing necessary Python packages ────────────────────────────────────────────────────────────────

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (22.0.2)
[... 182 lines cut ...]
auto-cpufreq version: 2.5.0

Python: 3.10.12
psutil package: 7.0.0
platform package: 1.0.8
click package: 8.1.8
distro package: 1.9.0

Computer type: Notebook
Docking Station
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.44 %

Total CPU usage: 1.0 %
Total system load: 0.78
Average temp. of all cores: 47.00 °C 

Currently using: performance governor
Currently turbo boost is: on

-------------------------------------------------------------------------------
```

AdnanHodzic, 2025-04-10:
I just made [another update](https://github.com/AdnanHodzic/auto-cpufreq/pull/826#issuecomment-2794549837), please get the latest changes `git pull`, followed by `sudo ./auto-cpufreq-installer --remove && sudo ./auto-cpufreq-installer --install`

noamtu123, 2025-04-10:
> [@noncarbonatedclack](https://github.com/noncarbonatedclack) & [@noamtu123](https://github.com/noamtu123) I just made [another update](https://github.com/AdnanHodzic/auto-cpufreq/pull/826#issuecomment-2794549837), please get the latest changes `git pull`, followed by `sudo ./auto-cpufreq-installer --remove && sudo ./auto-cpufreq-installer --install`

Still not working

AdnanHodzic, 2025-04-10:
When I run:

```
sudo ./auto-cpufreq-installer --install

────────────────── Detected Debian based distribution ────────────────

──────────────────────── Setting up Python environment ──────────────────────

Hit:1 http://nl.archive.ubuntu.com/ubuntu noble InRelease
Hit:2 http://nl.archive.ubuntu.com/ubuntu noble-updates InRelease                                                 
Hit:3 http://nl.archive.ubuntu.com/ubuntu noble-backports InRelease                                               
Hit:4 http://security.ubuntu.com/ubuntu noble-security InRelease                                                  
Hit:5 https://download.docker.com/linux/ubuntu noble InRelease                                                    
Hit:6 https://dl.google.com/linux/chrome/deb stable InRelease                                                     
Get:7 https://esm.ubuntu.com/apps/ubuntu noble-apps-security InRelease [7,595 B]
Hit:8 https://repo.protonvpn.com/debian stable InRelease   
Get:9 https://esm.ubuntu.com/apps/ubuntu noble-apps-updates InRelease [7,480 B]
Get:10 https://esm.ubuntu.com/infra/ubuntu noble-infra-security InRelease [7,474 B]
Get:11 https://esm.ubuntu.com/infra/ubuntu noble-infra-updates InRelease [7,473 B]
Fetched 30.0 kB in 2s (19.4 kB/s)     
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
18 packages can be upgraded. Run 'apt list --upgradable' to see them.
PyGObject version updated to ^3.50.0 in pyproject.toml
libgirepository-2.0-dev needs to be installed for version trixie/sid
---- 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.12.3-0ubuntu2).
python3-pip is already the newest version (24.0+dfsg-1ubuntu1.1).
python3-venv is already the newest version (3.12.3-0ubuntu2).
python3-setuptools is already the newest version (68.1.2-2ubuntu1.1).
dmidecode is already the newest version (3.5-3ubuntu0.1).
libgirepository-2.0-dev is already the newest version (2.80.0-6ubuntu3.2).
libcairo2-dev is already the newest version (1.18.0-3build1).
libgtk-3-dev is already the newest version (3.24.41-4ubuntu1.2).
gcc is already the newest version (4:13.2.0-7ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
─────────────────────── Installing necessary Python packages ─────────────────

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (25.0.1)
```

Right at the beginning I get:

```
PyGObject version updated to ^3.50.0 in pyproject.toml
libgirepository-2.0-dev needs to be installed for version trixie/sid
```

What do you get for these 2 lines?

noamtu123, 2025-04-10:
PyGObject version updated to 3.50.0 in pyproject.toml
libgirepository1.0-dev needs to be installed for version bookworm/sid

AdnanHodzic, 2025-04-10:
This is good:
```
PyGObject version updated to 3.50.0 in pyproject.toml
libgirepository1.0-dev needs to be installed for version bookworm/sid
```

But then this is the problem:

```
../meson.build:31:9: ERROR: Dependency 'girepository-2.0' is required but not found.
```

and I'm not sure where this is coming from. Only thing I can think of is that this is you install all of these locations:

```
rm -rf  auto-cpufreq/ build .mesonpy-* /tmp/pip-install-* /tmp/pip-req-build-*
```

Clone auto-cpufreq and install it again.

Alternatively, if you're still facing this problem and are not going to use GUI (because it doesn't work with Snap). I would just recommend you install with [Snap package ](https://github.com/AdnanHodzic/auto-cpufreq?tab=readme-ov-file#snap-store)which is not going to have any of these dependencies issues.

noamtu123, 2025-04-10:
> This is good:
> 
> ```
> PyGObject version updated to 3.50.0 in pyproject.toml
> libgirepository1.0-dev needs to be installed for version bookworm/sid
> ```
> 
> But then this is the problem:
> 
> ```
> ../meson.build:31:9: ERROR: Dependency 'girepository-2.0' is required but not found.
> ```
> 
> and I'm not sure where this is coming from. Only thing I can think of is that this is a leftover from first install so please try to remove everything under following locations to clean-up:
> 
> ```
> rm -rf  auto-cpufreq/ build .mesonpy-* /tmp/pip-install-* /tmp/pip-req-build-*
> ```
> 
> Clone auto-cpufreq and install it again.
> 
> Alternatively, if you're still facing this problem and are not going to use GUI (because it doesn't work with Snap). I would just recommend you install with [Snap package ](https://github.com/AdnanHodzic/auto-cpufreq?tab=readme-ov-file#snap-store)which is not going to have any of these dependencies issues.

well, i tried and it didn't help you can see there(what shown here is after i removed auto-cpufreq): 
```
noamtu123@noamtu123-Inspiron-5459:~$ rm -rf  auto-cpufreq/ build .mesonpy-* /tmp/pip-install-* /tmp/pip-req-build-*
noamtu123@noamtu123-Inspiron-5459:~$ git clone https://github.com/AdnanHodzic/auto-cpufreq.git
cd auto-cpufreq && sudo ./auto-cpufreq-installer
Cloning into 'auto-cpufreq'...
remote: Enumerating objects: 2841, done.
remote: Counting objects: 100% (910/910), done.
remote: Compressing objects: 100% (249/249), done.
remote: Total 2841 (delta 788), reused 664 (delta 661), pack-reused 1931 (from 2)
Receiving objects: 100% (2841/2841), 964.75 KiB | 1.83 MiB/s, done.
Resolving deltas: 100% (1587/1587), done.
[sudo] password for noamtu123:      

─────────────────────────── auto-cpufreq installer ───────────────────────────

Welcome to auto-cpufreq tool installer.

Select a key [I]nstall/[R]emove or press ctrl+c to quit: i

───────────────────── Detected Debian based distribution ─────────────────────

──────────────────────── Setting up Python environment ────────────────────────

Hit:1 http://il.archive.ubuntu.com/ubuntu jammy InRelease                      
Hit:2 http://il.archive.ubuntu.com/ubuntu jammy-updates InRelease              
Hit:3 http://security.ubuntu.com/ubuntu jammy-security InRelease               
Hit:4 http://il.archive.ubuntu.com/ubuntu jammy-backports InRelease            
Hit:5 https://ppa.launchpadcontent.net/zorinos/apps/ubuntu jammy InRelease     
Hit:6 https://ppa.launchpadcontent.net/zorinos/drivers/ubuntu jammy InRelease  
Hit:7 https://ppa.launchpadcontent.net/zorinos/patches/ubuntu jammy InRelease  
Hit:8 https://packages.mozilla.org/apt mozilla InRelease                       
Hit:9 https://ppa.launchpadcontent.net/zorinos/stable/ubuntu jammy InRelease
Hit:10 https://packages.zorinos.com/stable jammy InRelease                  
Hit:11 https://packages.zorinos.com/patches jammy InRelease
Hit:12 https://packages.zorinos.com/apps jammy InRelease
Hit:13 https://packages.zorinos.com/drivers jammy InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
PyGObject version updated to 3.50.0 in pyproject.toml
libgirepository1.0-dev needs to be installed for version bookworm/sid
---- 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
[... 77 lines cut ...]
Requirement already satisfied: urwid<3.0.0,>=2.6.16 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from auto-cpufreq==2.5.0+6789cb6) (2.6.16)
Requirement already satisfied: charset-normalizer<4,>=2 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from requests<3.0.0,>=2.32.3->auto-cpufreq==2.5.0+6789cb6) (3.4.1)
Requirement already satisfied: idna<4,>=2.5 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from requests<3.0.0,>=2.32.3->auto-cpufreq==2.5.0+6789cb6) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from requests<3.0.0,>=2.32.3->auto-cpufreq==2.5.0+6789cb6) (2.4.0)
Requirement already satisfied: certifi>=2017.4.17 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from requests<3.0.0,>=2.32.3->auto-cpufreq==2.5.0+6789cb6) (2025.1.31)
Requirement already satisfied: typing-extensions in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from urwid<3.0.0,>=2.6.16->auto-cpufreq==2.5.0+6789cb6) (4.13.2)
Requirement already satisfied: wcwidth in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from urwid<3.0.0,>=2.6.16->auto-cpufreq==2.5.0+6789cb6) (0.2.13)
Building wheels for collected packages: auto-cpufreq
  Building wheel for auto-cpufreq (pyproject.toml) ... done
  Created wheel for auto-cpufreq: filename=auto_cpufreq-2.5.0+6789cb6-py3-none-any.whl size=49137 sha256=eb7556d67c5c5bec537c8794d526ffb3530c1f0e131a2852a6a2746e46db76b8
  Stored in directory: /root/.cache/pip/wheels/ff/24/63/08867e8efba8de960065fe542e6a21683ab1eb0331354eb9ee
Successfully built auto-cpufreq
Installing collected packages: auto-cpufreq
  Attempting uninstall: auto-cpufreq
    Found existing installation: auto-cpufreq 2.5.0+6789cb6
    Uninstalling auto-cpufreq-2.5.0+6789cb6:
      Successfully uninstalled auto-cpufreq-2.5.0+6789cb6
Successfully installed auto-cpufreq-2.5.0+6789cb6

────────────────── auto-cpufreq tool successfully installed ──────────────────

For list of options, run:
auto-cpufreq --help

```

noncarbonatedclack, 2025-04-16:
I'm also still getting the error for ../meson.build:31:9: ERROR: Dependency 'girepository-2.0' is required but not found.

I have tried this, did not seem to change anything.
`rm -rf  auto-cpufreq/ build .mesonpy-* /tmp/pip-install-* /tmp/pip-req-build-*`

I've also removed the app entirely, deleted the folder, and pulled from git again. details below:

`───────────────────────────────────────────────────────────────────────── auto-cpufreq installer ─────────────────────────────────────────────────────────────────────────

Welcome to auto-cpufreq tool installer.

Select a key [I]nstall/[R]emove or press ctrl+c to quit: i

─────────────────────────────────────────────────────────────────── Detected Debian based distribution ───────────────────────────────────────────────────────────────────

────────────────────────────────────────────────────────────────────── Setting up Python environment ──────────────────────────────────────────────────────────────────────
                                     
Hit:3 http://apt.pop-os.org/proprietary jammy InRelease                                               
Hit:4 http://apt.pop-os.org/release jammy InRelease                             
Hit:6 http://apt.pop-os.org/ubuntu jammy InRelease                              
Hit:7 http://apt.pop-os.org/ubuntu jammy-security InRelease                     
Hit:8 http://apt.pop-os.org/ubuntu jammy-updates InRelease                      
Hit:9 http://apt.pop-os.org/ubuntu jammy-backports InRelease                    
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
PyGObject version updated to 3.50.0 in pyproject.toml
libgirepository1.0-dev needs to be installed for version bookworm/sid
---- 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gcc is already the newest version (4:11.2.0-1ubuntu1).
libcairo2-dev is already the newest version (1.16.0-5ubuntu2).
libgirepository1.0-dev is already the newest version (1.72.0-1).
libgtk-3-dev is already the newest version (3.24.33-1ubuntu2.2).
python3-setuptools is already the newest version (59.6.0-1.2ubuntu0.22.04.2).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.5).
dmidecode is already the newest version (3.3-3ubuntu0.2).
python3-dev is already the newest version (3.10.6-1~22.04.1).
python3-venv is already the newest version (3.10.6-1~22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

────────────────────────────────────────────────────────────────── Installing necessary Python packages ──────────────────────────────────────────────────────────────────

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (25.0.1)
Requirement already satisfied: wheel in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (0.45.1)
Collecting PyGObject
  Using cached pygobject-3.52.3.tar.gz (1.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      + meson setup /tmp/pip-install-03cga3pt/pygobject_b6cda7b77d944e6ca93f7c938ff11fe0 /tmp/pip-install-03cga3pt/pygobject_b6cda7b77d944e6ca93f7c938ff11fe0/.mesonpy-ix0ugp1q -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md -Dtests=false -Dwheel=true --wrap-mode=nofallback --native-file=/tmp/pip-install-03cga3pt/pygobject_b6cda7b77d944e6ca93f7c938ff11fe0/.mesonpy-ix0ugp1q/meson-python-native-file.ini
      The Meson build system
      Version: 1.7.2
      Source dir: /tmp/pip-install-03cga3pt/pygobject_b6cda7b77d944e6ca93f7c938ff11fe0
      Build dir: /tmp/pip-install-03cga3pt/pygobject_b6cda7b77d944e6ca93f7c938ff11fe0/.mesonpy-ix0ugp1q
      Build type: native build
      Project name: pygobject
      Project version: 3.52.3
      C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
      C linker for the host machine: cc ld.bfd 2.38
      Host machine cpu family: x86_64
[... 34 lines cut ...]
Requirement already satisfied: click<9.0.0,>=8.1.0 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from auto-cpufreq==2.5.0+b6493ed) (8.1.8)
Requirement already satisfied: distro<2.0.0,>=1.8.0 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from auto-cpufreq==2.5.0+b6493ed) (1.9.0)
Requirement already satisfied: psutil<8.0.0,>=6.0.0 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from auto-cpufreq==2.5.0+b6493ed) (7.0.0)
Requirement already satisfied: pyasyncore<2.0.0,>=1.0.4 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from auto-cpufreq==2.5.0+b6493ed) (1.0.4)
Requirement already satisfied: requests<3.0.0,>=2.32.3 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from auto-cpufreq==2.5.0+b6493ed) (2.32.3)
Requirement already satisfied: urwid<3.0.0,>=2.6.16 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from auto-cpufreq==2.5.0+b6493ed) (2.6.16)
Requirement already satisfied: charset-normalizer<4,>=2 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from requests<3.0.0,>=2.32.3->auto-cpufreq==2.5.0+b6493ed) (3.4.1)
Requirement already satisfied: idna<4,>=2.5 in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from requests<3.0.0,>=2.32.3->auto-cpufreq==2.5.0+b6493ed) (3.10)
[... cut ...]
  Stored in directory: /root/.cache/pip/wheels/cb/bd/e7/71ee5d8af7e65a334bca2734bb1c3f5cec38c7cc1eb5a46256
Successfully built auto-cpufreq
Installing collected packages: auto-cpufreq
  Attempting uninstall: auto-cpufreq
    Found existing installation: auto-cpufreq 2.5.0+b6493ed
    Uninstalling auto-cpufreq-2.5.0+b6493ed:
      Successfully uninstalled auto-cpufreq-2.5.0+b6493ed
Successfully installed auto-cpufreq-2.5.0+b6493ed

──────────────────────────────────────────────────────────────── auto-cpufreq tool successfully installed ────────────────────────────────────────────────────────────────`

gorbehnare, 2025-04-17:
> This is good:
>
> ```
> PyGObject version updated to 3.50.0 in pyproject.toml
> libgirepository1.0-dev needs to be installed for version bookworm/sid
> ```
>
> But then this is the problem:
>
> ``` [...]

On Kubuntu 25.04, I had this exact issue. I upgraded to 25.04 from 24.10 today (not a clean install), this fixed the issue on my system.

noamtu123, 2025-04-18:
**I tried again doing sudo apt update && sudo apt upgrade then i have done this:** 

```
noamtu123@noamtu123-Inspiron-5459:~$ rm -rf  auto-cpufreq/ build .mesonpy-* /tmp/pip-install-* /tmp/pip-req-build-*
noamtu123@noamtu123-Inspiron-5459:~$ git clone https://github.com/AdnanHodzic/auto-cpufreq.git
cd auto-cpufreq && sudo ./auto-cpufreq-installer
Cloning into 'auto-cpufreq'...
remote: Enumerating objects: 2882, done.
remote: Counting objects: 100% (953/953), done.
remote: Compressing objects: 100% (293/293), done.
remote: Total 2882 (delta 810), reused 679 (delta 660), pack-reused 1929 (from 2)
Receiving objects: 100% (2882/2882), 994.11 KiB | 2.25 MiB/s, done.
Resolving deltas: 100% (1607/1607), done.

─────────────────────────── auto-cpufreq installer ───────────────────────────

Welcome to auto-cpufreq tool installer.

Select a key [I]nstall/[R]emove or press ctrl+c to quit: i

───────────────────── Detected Debian based distribution ─────────────────────

──────────────────────── Setting up Python environment ────────────────────────

Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 http://il.archive.ubuntu.com/ubuntu jammy InRelease                      
Hit:3 http://il.archive.ubuntu.com/ubuntu jammy-updates InRelease              
Hit:4 https://packages.mozilla.org/apt mozilla InRelease                       
Hit:5 https://ppa.launchpadcontent.net/zorinos/apps/ubuntu jammy InRelease     
Hit:6 http://il.archive.ubuntu.com/ubuntu jammy-backports InRelease            
Hit:7 https://ppa.launchpadcontent.net/zorinos/drivers/ubuntu jammy InRelease
Hit:8 https://ppa.launchpadcontent.net/zorinos/patches/ubuntu jammy InRelease
Hit:9 https://ppa.launchpadcontent.net/zorinos/stable/ubuntu jammy InRelease
Hit:10 https://packages.zorinos.com/stable jammy InRelease
Hit:11 https://packages.zorinos.com/patches jammy InRelease
Hit:12 https://packages.zorinos.com/apps jammy InRelease
Hit:13 https://packages.zorinos.com/drivers jammy InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
PyGObject version updated to 3.50.0 in pyproject.toml
libgirepository1.0-dev needs to be installed for version bookworm/sid
---- 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gcc is already the newest version (4:11.2.0-1ubuntu1).
libcairo2-dev is already the newest version (1.16.0-5ubuntu2).
libgirepository1.0-dev is already the newest version (1.72.0-1).
dmidecode is already the newest version (3.3-3ubuntu0.2).
libgtk-3-dev is already the newest version (3.24.33-1ubuntu2.2).
python3-dev is already the newest version (3.10.6-1~22.04.1).
python3-setuptools is already the newest version (59.6.0-1.2ubuntu0.22.04.2).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.5).
python3-venv is already the newest version (3.10.6-1~22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

──────────────────── Installing necessary Python packages ────────────────────

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (25.0.1)
Requirement already satisfied: wheel in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (0.45.1)
Collecting PyGObject
  Using cached pygobject-3.52.3.tar.gz (1.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
[... 139 lines cut ...]

Python: 3.10.12
psutil package: 7.0.0
platform package: 1.0.8
click package: 8.1.8
distro package: 1.9.0

Computer type: Notebook
Docking Station
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.44 %

Total CPU usage: 1.3 %
Total system load: 1.38
Average temp. of all cores: 27.00 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

```

AdnanHodzic, 2025-04-18:
> On Kubuntu 25.04, I had this exact issue. I upgraded to 25.04 from 24.10 today (not a clean install), this fixed the issue on my system.

auto-cpufreq has nothing to do with dist upgrade. Please run what was suggested here.

noamtu123, 2025-04-18:
Alreqdy did nothing working

noncarbonatedclack, 2025-04-19:
It seems that this error is something both noamtu123 and I are getting, is there a way we can enable use of fallback dependencies on our own? 

```      
      Run-time dependency girepository-2.0 found: NO (tried pkgconfig and cmake)
      Not looking for a fallback subproject for the dependency girepository-2.0 because:
      Use of fallback dependencies is disabled.```

AdnanHodzic, 2025-04-19:
> Related to improvements made as part of [#826](https://github.com/AdnanHodzic/auto-cpufreq/pull/826).
> 
> [@kirmola](https://github.com/kirmola) in this case it's trying to install `libgirepository-2.0-dev` when it shouldn't as it won't find it for 22.04 repos.

@noncarbonatedclack it has all been said, and as I mentioned this goes back to #826. Since then I made [couple of changes](https://github.com/AdnanHodzic/auto-cpufreq/pull/826#issuecomment-2794549837) that should've fixed this issue, but apparently they didn't on all Debian/Ubuntu based derivatives. 

[Commit 6789cb6](https://github.com/AdnanHodzic/auto-cpufreq/commit/6789cb6ce48a741dbd881dc3fc5fcd12250dc876) explains the whole logic.

So if you commented out lines line [87 to 89 of auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto-cpufreq-installer#L87), and [turned `else` in line 90 to `if`](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto-cpufreq-installer#L90) it should only install `libgirepository1.0-dev` package.

Since I'm limited with available time to do this myself and [project is looking for contributors](https://github.com/AdnanHodzic/auto-cpufreq#looking-for-developers-and-co-maintainers) if you or anyone else wants to give it a try in fixing or further improving dependency management, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).