AdnanHodzic/auto-cpufreq #842: /usr/bin/cpufreqctl.auto-cpufreq: line 99: ec ho: write error: Device or resource busy

itzayush69, 2025-05-04:
> I'm on Arch Linux on a Lenovo IdeaPad Slim 5 with an Intel Core Ultra 5 125H CPU, and I still get this error when in charging mode.
> 
> Exact error:
> ```
> /usr/bin/cpufreqctl.auto-cpufreq: line 99: echo: write error: Device or resource busy
> ``` 

 _Originally posted by @itzayush69 in [#751](https://github.com/AdnanHodzic/auto-cpufreq/issues/751#issuecomment-2849155666)_

[image]

**Note:** This error only occurs when the laptop is connected to AC power (plugged in).

Below is the config (```/etc/auto-cpufreq.conf```):
_I haven't changed anything, just added `ideapad_laptop_conservation_mode = true`_
```
# settings for when connected to a power source
[charger]
# see available governors by running: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
# preferred governor
governor = performance

# EPP: see available preferences by running: cat /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences
energy_performance_preference = performance

# EPB (Energy Performance Bias) for the intel_pstate driver
# see conversion info: https://www.kernel.org/doc/html/latest/admin-guide/pm/intel_epb.html
# available EPB options include a numeric value between 0-15
# (where 0 = maximum performance and 15 = maximum power saving),
# or one of the following strings:
# performance (0), balance_performance (4), default (6), balance_power (8), or power (15)
# if the parameter is missing in the config and the hardware supports this setting, the default value will be used
# the default value is `balance_performance` (for charger)
# energy_perf_bias = balance_performance
energy_perf_bias = performance

# Platform Profiles
# https://www.kernel.org/doc/html/latest/userspace-api/sysfs-platform_profile.html
# See available options by running:
# cat /sys/firmware/acpi/platform_profile_choices
# platform_profile = performance

# minimum cpu frequency (in kHz)
# example: for 800 MHz = 800000 kHz --> scaling_min_freq = 800000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_min_freq = 800000

# maximum cpu frequency (in kHz)
# example: for 1GHz = 1000 MHz = 1000000 kHz -> scaling_max_freq = 1000000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_max_freq = 1000000

# turbo boost setting. possible values: always, auto, never
turbo = auto

# settings for when using battery power
[battery]
# see available governors by running: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
# preferred governor
governor = powersave

# EPP: see available preferences by running: cat /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences
energy_performance_preference = power

# EPB (Energy Performance Bias) for the intel_pstate driver
# see conversion info: https://www.kernel.org/doc/html/latest/admin-guide/pm/intel_epb.html
# available EPB options include a numeric value between 0-15
# (where 0 = maximum performance and 15 = maximum power saving),
# or one of the following strings:
# performance (0), balance_performance (4), default (6), balance_power (8), or power (15)
# if the parameter is missing in the config and the hardware supports this setting, the default value will be used
# the default value is `balance_power` (for battery)
# energy_perf_bias = balance_power

# Platform Profiles
# https://www.kernel.org/doc/html/latest/userspace-api/sysfs-platform_profile.html
# See available options by running:
# cat /sys/firmware/acpi/platform_profile_choices
# platform_profile = low-power

# minimum cpu frequency (in kHz)
# example: for 800 MHz = 800000 kHz --> scaling_min_freq = 800000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_min_freq = 800000

# maximum cpu frequency (in kHz)
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# example: for 1GHz = 1000 MHz = 1000000 kHz -> scaling_max_freq = 1000000
# to use this feature, uncomment the following line and set the value accordingly
# scaling_max_freq = 1000000

# turbo boost setting (always, auto, or never)
turbo = auto

# battery charging threshold
# reference: https://github.com/AdnanHodzic/auto-cpufreq/#battery-charging-thresholds
#enable_thresholds = true
#start_threshold = 20
#stop_threshold = 80

ideapad_laptop_conservation_mode = true
```

PurpleWazard, 2025-05-10:
Do you still have this issue when ideapad_laptop_conservation_mode = true ?

itzayush69, 2025-05-10:
> [@itzayush69](https://github.com/itzayush69) you still have this issue when ideapad_laptop_conservation_mode = false ?

Yeah, the issue persists even with ideapad_laptop_conservation_mode set to false, or even when completely removing that line.

MixedOne, 2025-06-14:
Hello, I can confirm  I  have the same problem using NixOS (doesn't mater if it's stable or unstable pkgs branch), it's present since a few months.
I'm keen to help to resolve the issue, please let me know if sharing any specific details would help.

w-A-L-L-e, 2025-06-17:
Confirmed. Also have exact same issue with asus zenbook (intel core ultra 9) on ubuntu 22.04 LTS:

```
2025-06-17T23:47:08.878061+02:00 wschrep-zenbook python[9817]: /usr/local/bin/cpufreqctl.auto-cpufreq: line 99: echo: write error: Device or resource busy
```
When unplugging power it stops. Luckily thereby the battery is not drained. Still I'd like it not to spam my syslog with errors whenever I charge. Is there some config setting we can alter to make this stop?

AdnanHodzic, 2026-04-19:
> This issue is caused by setting the Governor to Performance which causes the system to lock the EPP settings (and corresponding) file. Powersave is the best setting for the governor - the name is a misnomer it just allows the EPP settings to work where the Performance governor mode hard-locks / over-rides the EPP settings. NO downside to leaving governor on Powersave mode all the time. See my response to this issue where I shared some of the AI explanatory notes on this one: EPP write fails with "Device or resource busy" on Intel Core Ultra 7 155H (Arch) [#927](https://github.com/AdnanHodzic/auto-cpufreq/issues/927)

@lianthar since [in comments](https://github.com/AdnanHodzic/auto-cpufreq/issues/927#issuecomment-4275102404) I see your suggestion is helping folks in #927. 

If you have ideas how to fix this issue, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

AdnanHodzic, 2026-08-19:
Changes were made as part of #958 which could've solved this issue, please pull the latest changes and re-install auto-cpufreq.

I'm closing the issue, if the problem persists, please comment or feel free to-reopen it. Thanks!

closed 2026-08-19 by AdnanHodzic