AdnanHodzic/auto-cpufreq #844: Installation failure from auto-cpufreq-installer

carver, 2025-05-18:
Note: I visited the [auto-cpufreq-genAI-chatbot](https://foolcontrol.org/?p=4903), but I'm getting:
```
Looks like there’s a problem with this site

https://foolcontrol.org/?p=4903 might have a temporary problem or it could have moved.

Error code: 500 Internal Server Error

    The site could be temporarily unavailable or too busy. Try again in a few moments.
```

### Have you tried?

- [x] [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- [ ] If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?

I was trying this installation method because the Snap package was spewing syslog errors about AppArmor denying permissions, and I saw a thread that said this other install was preferred.

### Error output:
```text
sudo ./auto-cpufreq-installer 

─────────────────────────────────────────────────────────────────────────────────────────────── auto-cpufreq installer ───────────────────────────────────────────────────────────────────────────────────────────────

Welcome to auto-cpufreq tool installer.

Select a key [I]nstall/[R]emove or press ctrl+c to quit: i

───────────────────────────────────────────────────────────────────────────────────────── Detected Debian based distribution ─────────────────────────────────────────────────────────────────────────────────────────

──────────────────────────────────────────────────────────────────────────────────────────── Setting up Python environment ────────────────────────────────────────────────────────────────────────────────────────────

Hit:1 https://packages.microsoft.com/repos/code stable InRelease
Hit:2 https://updates.signal.org/desktop/apt xenial InRelease                     
Hit:3 http://us.archive.ubuntu.com/ubuntu noble InRelease                         
Hit:4 http://us.archive.ubuntu.com/ubuntu noble-updates InRelease            
Hit:5 http://us.archive.ubuntu.com/ubuntu noble-backports InRelease          
Hit:6 http://security.ubuntu.com/ubuntu noble-security InRelease             
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
PyGObject version updated to ^3.50.0 in pyproject.toml
libgirepository-2.0-dev needs to be installed for version trixie/sid
---- 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.12.3-0ubuntu2).
python3-pip is already the newest version (24.0+dfsg-1ubuntu1.1).
python3-venv is already the newest version (3.12.3-0ubuntu2).
python3-setuptools is already the newest version (68.1.2-2ubuntu1.1).
dmidecode is already the newest version (3.5-3ubuntu0.1).
libgirepository-2.0-dev is already the newest version (2.80.0-6ubuntu3.2).
libcairo2-dev is already the newest version (1.18.0-3build1).
libgtk-3-dev is already the newest version (3.24.41-4ubuntu1.3).
gcc is already the newest version (4:13.2.0-7ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

──────────────────────────────────────────────────────────────────────────────────────── Installing necessary Python packages ────────────────────────────────────────────────────────────────────────────────────────

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (25.1.1)
Requirement already satisfied: wheel in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (0.45.1)
Requirement already satisfied: PyGObject in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (3.52.3)
Requirement already satisfied: pycairo>=1.16 in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from PyGObject) (1.28.0)

──────────────────────────────────────────────────────────────────────────────────────────── Installing auto-cpufreq tool ────────────────────────────────────────────────────────────────────────────────────────────

Processing /home/carver/code/auto-cpufreq
  Installing build dependencies ... done
[... 89 lines cut ...]

────────────────────────────────────────────────────────────────────────────────────── auto-cpufreq tool successfully installed ──────────────────────────────────────────────────────────────────────────────────────

For list of options, run:
auto-cpufreq --help

carver@carver-framework-2023:~/code/auto-cpufreq$ which python3.12
/usr/bin/python3.12
carver@carver-framework-2023:~/code/auto-cpufreq$ python3.12
Python 3.12.3 (main, Feb  4 2025, 14:48:35) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
carver@carver-framework-2023:~/code/auto-cpufreq$ auto-cpufreq --help
/opt/auto-cpufreq/venv/bin/python: can't open file '/opt/auto-cpufreq/venv/bin/auto-cpufreq': [Errno 2] No such file or directory
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
/opt/auto-cpufreq/venv/bin/python: can't open file '/opt/auto-cpufreq/venv/bin/auto-cpufreq': [Errno 2] No such file or directory
```

AdnanHodzic, 2025-05-18:
> Note: I visited the [auto-cpufreq-genAI-chatbot](https://foolcontrol.org/?p=4903), but I'm getting:
>
> Looks like there’s a problem with this site
>
> https://foolcontrol.org/?p=4903 might have a [...]

Do you still get this issue? As site was (and still is) up and the are no problems with opening ["How I replaced myself with a genAI chatbot using Gemini"](https://foolcontrol.org/?p=4903).

I suggest to re-install auto-cpufreq, e.g: 

`sudo ./auto-cpufreq-installer --remove`

Clean up build files, since I see it's trying to use a lot of cached libraries:

```
rm -rf /tmp/pip-build-env-*
rm -rf /tmp/pip-install-*
```

Make sure you have the latest code and [try to install it again](https://github.com/AdnanHodzic/auto-cpufreq/?tab=readme-ov-file#auto-cpufreq-installer).

Regarding snap package:
> I was trying this installation method because the Snap package was spewing syslog errors about AppArmor denying permissions, and I saw a thread that said this other install was preferred.

If you follow the [snap installation](https://github.com/AdnanHodzic/auto-cpufreq/?tab=readme-ov-file#snap-store), it should all work fine, only problem is that GUI won't work due to Snap restrictions.

carver, 2025-05-19:
> Do you still get this issue? As site was (and still is) up and the are no problems with opening ["How I replaced myself with a genAI chatbot using Gemini"](https://foolcontrol.org/?p=4903).

Ah, bummer. It loads for me now. It was an intermittent problem, apparently.

> If you follow the [snap installation](https://github.com/AdnanHodzic/auto-cpufreq/?tab=readme-ov-file#snap-store), it should all work fine, only problem is that GUI won't work due to Snap restrictions.

Unfortunately, it doesn't seem to be working. I tried removing and reinstalling with snap. I continue to get permission errors in my syslog like:
```
2025-05-19T11:39:36.096547-07:00 carver-framework-2023 kernel: audit: type=1400 audit(1747679976.095:2459119): apparmor="DENIED" operation="open" class="file" profile="snap.auto-cpufreq.service" name="/sys/devices/system/cpu/cpu9/power/energy_perf_bias" pid=1420915 comm="bash" requested_mask="wc" denied_mask="wc" fsuid=0 ouid=0
2025-05-19T11:39:36.097704-07:00 carver-framework-2023 auto-cpufreq.service[1420915]: /snap/auto-cpufreq/160/usr/bin/cpufreqctl.auto-cpufreq: line 288: /sys/devices/system/cpu/cpu10/power/energy_perf_bias: Permission denied
2025-05-19T11:39:36.099975-07:00 carver-framework-2023 auto-cpufreq.service[1420915]: /snap/auto-cpufreq/160/usr/bin/cpufreqctl.auto-cpufreq: line 288: /sys/devices/system/cpu/cpu11/power/energy_perf_bias: Permission denied
2025-05-19T11:39:36.101753-07:00 carver-framework-2023 auto-cpufreq.service[1420915]: /snap/auto-cpufreq/160/usr/bin/cpufreqctl.auto-cpufreq: line 288: /sys/devices/system/cpu/cpu12/power/energy_perf_bias: Permission denied
... (lots more)
```

> Maybe even a good idea to remove old auto-cpufreq repo code and [try to install it again](https://github.com/AdnanHodzic/auto-cpufreq/?tab=readme-ov-file#auto-cpufreq-installer).

Ok, I tried this (plus nuking pip cache files, etc, that you suggested). I wish I understood why, but that does seem to do the trick. I am up and running now. Thanks!

I'm happy to answer questions or try things if you're curious about why my snap install is giving permissions errors. Otherwise, I'm all set if you want to close. 👍🏻

AdnanHodzic, 2025-05-20:
> Ok, I tried this (plus nuking pip cache files, etc, that you suggested). I wish I understood why, but that does seem to do the trick. I am up and running now. Thanks!

It was using (old) cached version of libraries. 

> I'm happy to answer questions or try things if you're curious about why my snap install is giving permissions errors. Otherwise, I'm all set if you want to close. 👍🏻

No need, as auto-cpufreq-intsaller is preferred installation method.  

Regardless, good to hear and I'm closing the issue, please consider donating or contributing to the project: https://github.com/AdnanHodzic/auto-cpufreq/#donate

closed 2025-05-20 by AdnanHodzic