AdnanHodzic/auto-cpufreq #845: Stay in battery mode even when charging

jack5766, 2025-05-23:
Hi,

First of all thank you for your auto-cpufreq tool.
I have a problem getting it to work.
When my computer is charging, the battery configuration is always applied.
I placed the configuration file in /etc but nothing changes.
My laptop is a CLEVO NS50MU.

Thank you for your help.

### Error output:
```text
No error output
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug

Using settings defined in /etc/auto-cpufreq.conf file

-------------------------------------------------------------------------------

Linux distro: Fedora Linux 42 Workstation Edition
Linux kernel: 6.14.6-300.fc42.x86_64
Processor: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2800 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0      4.0%        40 °C       768 MHz
CPU1      4.9%        42 °C       542 MHz
CPU2      4.0%        38 °C      1300 MHz
CPU3      0.0%        39 °C       400 MHz
CPU4      3.9%        40 °C       400 MHz
CPU5      0.0%        42 °C       400 MHz
CPU6      1.0%        38 °C      1300 MHz
CPU7      5.9%        39 °C      1300 MHz

auto-cpufreq version: 2.6.0

Python: 3.13.3
psutil package: 7.0.0
platform package: 1.0.8
click package: 8.2.0
distro package: 1.9.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.07 %

Total CPU usage: 1.4 %
Total system load: 0.34
Average temp. of all cores: 39.00 °C 

Currently using: powersave governor
Currently turbo boost is: off

```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

jack5766, 2025-05-26:
Hi,
If I don't use the configuration file, the problem remains the same.

zenzer0s, 2025-05-31:
try to disable intel pstate and check they might be conflicated

give a try to install from git version