AdnanHodzic/auto-cpufreq #848: Bug on the battery detection (Ubuntu : 24.04) -> fixed on the #847

baldeuniversel, 2025-05-27:
### Error output:
```text
Traceback (most recent call last):
  File "/opt/auto-cpufreq/venv/bin/auto-cpufreq", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/click/core.py", line 1442, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/click/core.py", line 1363, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/click/core.py", line 1226, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/click/core.py", line 794, in invoke
    return callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/bin/auto_cpufreq.py", line 236, in main
    monitor.run()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_monitor.py", line 297, in run
    self.loop.run()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/main_loop.py", line 337, in run
    self._run()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/main_loop.py", line 439, in _run
    self.event_loop.run()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/select_loop.py", line 182, in run
    self._loop()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/select_loop.py", line 224, in _loop
    alarm_callback()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/main_loop.py", line 218, in cb
    callback(self, user_data)
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_monitor.py", line 98, in update
    report: SystemReport = system_info.generate_system_report()
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_info.py", line 333, in generate_system_report
    battery_info=self.battery_info(),
                 ^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_info.py", line 263, in battery_info
    energy_rate: str | None = read_file(
                              ^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_info.py", line 221, in read_file
    return f.read().strip()
           ^^^^^^^^
OSError: [Errno 19] No such device
```
---

### System information:

```
sudo auto-cpufreq --stats
Traceback (most recent call last):
  File "/opt/auto-cpufreq/venv/bin/auto-cpufreq", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/click/core.py", line 1442, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/click/core.py", line 1363, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/click/core.py", line 1226, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/click/core.py", line 794, in invoke
    return callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/bin/auto_cpufreq.py", line 236, in main
    monitor.run()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_monitor.py", line 297, in run
    self.loop.run()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/main_loop.py", line 337, in run
    self._run()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/main_loop.py", line 439, in _run
    self.event_loop.run()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/select_loop.py", line 182, in run
    self._loop()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/select_loop.py", line 224, in _loop
    alarm_callback()
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/urwid/event_loop/main_loop.py", line 218, in cb
    callback(self, user_data)
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_monitor.py", line 98, in update
    report: SystemReport = system_info.generate_system_report()
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_info.py", line 333, in generate_system_report
    battery_info=self.battery_info(),
                 ^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_info.py", line 263, in battery_info
    energy_rate: str | None = read_file(
                              ^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq/modules/system_info.py", line 221, in read_file
    return f.read().strip()
           ^^^^^^^^
OSError: [Errno 19] No such device
```

---

### Bug fixed 
Bug fixed on the commit #847

AdnanHodzic, 2025-05-27:
#847 has been merged with `master` brach.

Please make sure to pull the latest changes and re-install auto-cpufreq using auto-cpufreq-installer

AdnanHodzic, 2025-05-29:
Closing the issue since #847 has been merged with `master` branch.

closed 2025-05-29 by AdnanHodzic